gnuradio/gnuradio #2508: Linking of libgnuradio-runtime fails with -DENABLE_INTERNAL_VOLK=OFF

rear1019, 2019-05-25:
Error message:
```
[100%] Linking CXX shared library libgnuradio-runtime-3.8git.so
/usr/bin/ld: cannot find -lVolk::volk
```

cmake command:
```
cmake -DCMAKE_BUILD_TYPE=Debug -DENABLE_INTERNAL_VOLK=OFF ..
```

External volk is found according to cmake output:
```
-- Configuring VOLK support...
-- Found PkgConfig: /usr/bin/pkg-config (found version "1.6.1") 
-- Checking for module 'volk'
--   Found volk, version 1.4
-- Found VOLK: /lib/libvolk.so  
--   An external VOLK has been found and will be used for build.
```

git bisect indicates that the error is introduced by commit ab2fb35677.

michaelld, 2019-05-25:
The current GNU Radio GIT master HEAD's CMake scripts are designed to work with the current Volk GIT master HEAD. You're trying to use the latest Volk release, which this version of GNU Radio isn't designed to work with. Yes, at issue is the relatively recent move to "modern CMake", for both GR and Volk. I have a PR in to fix the Volk side https://github.com/gnuradio/volk/pull/248 , and I'm working on the GR side too to restore certain pre-modern CMake functionality and compatibility. Hopefully I'll get a PR in for that next week. Until then, if you can move to either the GR-internal Volk or the current Volk GIT master HEAD, those should work for your needs.

noc0lour, 2019-05-26:
Hm, should update to find VOLK 1.5 since that will be a compatible release again.

michaelld, 2019-05-26:
> Hm, should update to find VOLK 1.5 since that will be a compatible release again.

Sure that's another option. Be timely, too, given all the changes and fixes in Volk.

Do you know if the latest GR _release_ would work properly with a new Volk release? I'm guessing it won't without my PR to bring back old-style variable usage ... but, I haven't looked into it yet.

NeuroForLunch, 2021-06-24:
It seems this issue was resolved. If so, please close it.

closed 2021-09-06 by rear1019