gnuradio/gnuradio #2510: pip install process fails under CentOS and Debian, same error.

NephtonLabs, 2019-05-27:
Following the guide here: https://wiki.gnuradio.org/index.php/InstallingGR under Using PyBOMBS
At the step
> pybombs prefix init ~/gnuradio -R gnuradio-default
I get a significant error. Console error, and logs referenced in error are  attached to the case. In short I get this: 

-- Boost include directories: /usr/include
-- Boost library directories: /usr/lib/x86_64-linux-gnu
-- Boost libraries: /usr/lib/x86_64-linux-gnu/libboost_chrono.so;/usr/lib/x86_64-linux-gnu/libboost_date_time.so;/usr/lib/x86_64-linux-gnu/libboost_filesystem.so;/usr/lib/x86_64-linux-gnu/libboost_program_options.so;/usr/lib/x86_64-linux-gnu/libboost_regex.so;/usr/lib/x86_64-linux-gnu/libboost_system.so;/usr/lib/x86_64-linux-gnu/libboost_unit_test_framework.so;/usr/lib/x86_64-linux-gnu/libboost_serialization.so;/usr/lib/x86_64-linux-gnu/libboost_thread.so;/usr/lib/x86_64-linux-gnu/libboost_atomic.so;/usr/lib/x86_64-linux-gnu/libpthread.so
-- 
-- Python checking for Python version 2.7 or greater
-- Python checking for Python version 2.7 or greater - found
-- 
-- Python checking for Mako templates 0.4.2 or greater
-- Python checking for Mako templates 0.4.2 or greater - "import mako" failed
-- 
-- Python checking for requests 2.0 or greater
-- Python checking for requests 2.0 or greater - found
-- 
-- Python checking for numpy 1.7 or greater
-- Python checking for numpy 1.7 or greater - "import numpy" failed
-- 
-- Configuring LibUHD support...
--   Dependency Boost_FOUND = 1
--   Dependency HAVE_PYTHON_PLAT_MIN_VERSION = TRUE
--   Dependency HAVE_PYTHON_MODULE_MAKO = FALSE
CMake Error at cmake/Modules/UHDComponent.cmake:59 (message):
  Dependencies for required component LibUHD not met.
Call Stack (most recent call first):
  CMakeLists.txt:392 (LIBUHD_REGISTER_COMPONENT)

-- Configuring incomplete, errors occurred!
See also "/home/user/gnuradio/src/uhd/host/build/CMakeFiles/CMakeOutput.log".
See also "/home/user/gnuradio/src/uhd/host/build/CMakeFiles/CMakeError.log".
PyBOMBS.Packager.source - ERROR - Configuration failed after running at least twice.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package uhd:
Configuration failed
PyBOMBS.install_manager - ERROR - Error installing package uhd. Aborting.

It should be noted that I've run

> pip install mako
and 

> pip install numpy 
Not sure why it isn't detecting them.
There is also an odd flag error in the CMakeError.log that happens in both Debian and CentOS. At this point I'm pretty sure there is an issue with the installer. 
[CMakeOutput.log](https://github.com/gnuradio/gnuradio/files/3225127/CMakeOutput.log)
[consoleError.txt](https://github.com/gnuradio/gnuradio/files/3225128/consoleError.txt)
[CMakeError.log](https://github.com/gnuradio/gnuradio/files/3225129/CMakeError.log)

noc0lour, 2019-05-28:
In both Debian & CentOS there should be `python{2,3}-mako` in the package sources. If possible please use the system packages.

closed 2019-05-28 by noc0lour

kdtmac, 2019-11-06:
same error. How to fix this? I m sure i installed mako and all modules. Maybe not with apt, because apt does not have.

velichkov, 2019-11-06:
Hi @kdtmac,

> same error. 

Please provide the exact errors you are getting.

> How to fix this? 

What Linux distribution do you use and what version?

> I m sure i installed mako and all modules. Maybe not with apt, because apt does not have.

Which pip version did you use to install pybombs and which pybombs version did you installed?
How did you installed mako, which version and for which python version?