gnuradio/gnuradio #2515: CMake using absolute path for installation of runtime_swig-release

flynn378, 2019-05-28:
During cross compiling the absolute path is saved as part of the file created by cmake:
 runtime_swigTargets-release.cmake
GNU Radio will build, install and execute with these settings, but OOT modules will not build.  
Based on the bug report:
https://cmake.org/Bug/print_bug_page.php?bug_id=14367
Which is marked as not a bug, but explains the issue very well. 
gnuradio-runtime.swig/CMakeLists.txt uses an absolute path for the DESTINATION ( or something that gets turned into an absolute path) in the lines:
install(
  TARGETS runtime_swig
  EXPORT runtime_swig-export
  DESTINATION ${GR_PYTHON_DIR}/gnuradio/gr
  ) 

The DESTINATION should point to a relative path, based on the above bug report.

There is an email chain on the mailing list with more information about the issue.

michaelld, 2019-05-29:
Should be fixed now : https://github.com/gnuradio/gnuradio/pull/2502

michaelld, 2019-05-29:
This PR was merged today. It changes the runtime_swig `DESTINATION` to `${GR_CMAKE_DIR}`, which by default is a relative path.

flynn378, 2019-05-29:
I will test this tomorrow, I believe the actual error is a relative path needed for the 
install(
TARGETS ...
${GR_PYTHON_DIR}/gnuradio/gr
lines 62-65
Not the cmake install directory which was changed in merge request #2502.  I think I tried that change in the days of testing.  I could not find anything which provides the relative Python Path, It has been added to CMake, I think 3.13 or 3.14.
If there is a variable which is the relative Python Path it would be useful.  
I have tested the following changes and they work,  I am not sure it is the correct solution.
Based on the information from:
https://stackoverflow.com/questions/1242904/finding-python-site-packages-directory-with-cmake

I modified GrPython.cmake by adding a new variable, PYTHON_SITE:
########################################################################
# Sets the python relative installation directory PYTHON_SITE
########################################################################
if(NOT DEFINED PYTHON_SITE)
execute_process(
  COMMAND "${PYTHON_EXECUTABLE}" -c "if True:
    from distutils import sysconfig as sc
    print(sc.get_python_lib(prefix='', plat_specific=True))"
  OUTPUT_VARIABLE PYTHON_SITE
  OUTPUT_STRIP_TRAILING_WHITESPACE)
endif()

and then changed the target install line 65 in 
swig/CmakeLists.txt to use PYTHON_SITE,

With these changes I can confirm GNU Radio builds and OOTs will build against this using both Yocto and local computer building.  I would like more testing on this solution and I am not sure it is the correct solution, but it is a working solution.

michaelld, 2019-05-29:
OK; interesting. Please do put together a PR for us to test when you get to that point!

noc0lour, 2019-07-17:
Manually closing this. Fix is in-tree.

michaelld, 2019-07-18:
@flynn378 Your change was merged, but it breaks installs that specify where to install Python. We're going to reset just the install setting, leaving the rest of the background code in place; here's the PR < https://github.com/gnuradio/gnuradio/pull/2640 >. If you want to fix the install path to be relative, that's great ... but, it needs to be for `GR_PYTHON_DIR`, whether determined internally or set externally. Can you come up with a new PR that removes the `GR_PYTHON_RELATIVE` creation, and instead does the same thing for the overall `GR_PYTHON_DIR` ... and then test / verify that it works for your needs. Once we have that we can test / verify on the CI as well as for package managers such as what I maintain that specify the directory ... we want all of these to work correctly! I know CMake can do this, it's a matter of putting the fixes in place & getting them tested. Thx!

marcusmueller, 2019-07-18:
I fully second this. Your efforts are highly appreciated! But your patch broke one of the largest distros, and a few other systems, so we had to reverse it in its core.

Please don't let that discourage you; you can borrow the Dockerfiles from the gnuradio/gnuradio-buildbot repo (and remove all the buildbot stuff) to whip up several Linux distros fully equipped to build GNU Radio right away. If in doubt, you can use that to test build and installation :)

maitbot, 2020-11-15:
With any luck this is also fixed by:
allow oot override of gr python dir #3944

marcusmueller, 2024-01-03:
SWIG-based GR is EOL

closed 2024-01-03 by marcusmueller