gnuradio/gnuradio #2518: qtgui time sink complex message input broken under grc

ChromeBumper, 2019-05-31:
If I connect to a qtgui time sink in grc it shows the input port as "in0" but the cc code declares the message port as "in"
short term options are to either modify time_sink_c_impl.cc to accept "in0" as:
      // setup PDU handling input port
//ddd 
//      message_port_register_in(pmt::mp("in"));
//      set_msg_handler(pmt::mp("in"),
//                      boost::bind(&time_sink_c_impl::handle_pdus, this, _1));
      message_port_register_in(pmt::mp("in0"));
      set_msg_handler(pmt::mp("in0"),
                      boost::bind(&time_sink_c_impl::handle_pdus, this, _1));

or to modify the python code that grc generates (every time) 
#        self.msg_connect((self.dave_specsplit_0, 'AngleVsFreq'), (self.qtgui_time_sink_x_0_0, 'in0'))
#change to:
        self.msg_connect((self.dave_specsplit_0, 'AngleVsFreq'), (self.qtgui_time_sink_x_0_0, 'in'))

dl1ksv, 2019-06-05:
in0 is the stream port and in is the message port, but the yml definition does not contain a message port definit ion as for instance the frequency sink does.

marcusmueller, 2024-01-03:
Fixed by #2543

closed 2024-01-03 by marcusmueller