gnuradio/gnuradio #2519: GRC doesn't generate valid C++ string parameters unless double-quotes are explicitly added

ncorgan, 2019-06-01:
When a string parameter is empty or does not have quotes (as is the case for many default parameter values), the generated C++ string is surrounded by single quotes, which is valid Python but not C++. If the string is given with single-quotes, the generated C++ string will have single-quotes. The only time a valid C++ string is generated is if the value itself is already surrounded by double-quotes.

[Attached](https://github.com/gnuradio/gnuradio/files/3243256/cpp_string_bug.txt) is a minimal .grc file (with the extension changed to .txt so Github will accept it) that demonstrates the behavior as of 27dd99e49a7d34eeb4c0a8fedfe3b072ad582662 under the following conditions:

Condition|Python Output|C++ Output
-------------|----------|----------
String parameter with no quotes|String in single quotes (valid)|String in single quotes **(invalid)**
String parameter in single quotes|String in single quotes (valid)|String in single quotes **(invalid)**
String parameter is empty|String in single quotes (valid)|String in single quotes **(invalid)**
String parameter is set to empty single-quotes|String in single quotes (valid)|String in single quotes **(invalid)**
String parameter is set to empty double-quotes|String in double quotes (valid)|String in double quotes(valid)
String parameter in double quotes|String in double quotes (valid)|String in double quotes(valid)

AsriFox, 2023-09-18:
The issue is still present in GNU Radio 3.10.7.

Some blocks use templates to work around this:
- in [audio::sink](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-audio/grc/audio_sink.block.yml#L48) and [audio::source](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-audio/grc/audio_source.block.yml#L48), the `"'": '"'` translation is used;
- in [qtgui::waterfall_sink_x](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-qtgui/grc/qtgui_waterfall_sink_x.block.yml#L329), the `"${no_quotes(name)}"` pattern is used;
  - `"${no_quotes(file)}"` is also used in [blocks::file_sink](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-blocks/grc/blocks_file_sink.block.yml#L55) with `dtype: file_save`, but not in [blocks::file_source](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-blocks/grc/blocks_file_source.block.yml#L63), for some reason.

Blocks that don't have workarounds:
- [parameter](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/grc/blocks/parameter.block.yml#L42-L43)
- [digital::correlate_access_code_xx_ts](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-digital/grc/digital_correlate_access_code_xx_ts.block.yml#L42-L45) (access_code, tagname)
- [digital::additive_scrambler_bb](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-digital/grc/digital_additive_scrambler_bb.block.yml#L54) (reset_tag_key)
- [blocks::tagged_stream_multiply_length](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-blocks/grc/blocks_tagged_stream_multiply_length.block.yml#L50), [blocks::tagged_stream_mux](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-blocks/grc/blocks_tagged_stream_mux.block.yml#L50), [blocks::tagged_stream_align](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-blocks/grc/blocks_tagged_stream_align.block.yml#L41) (lengthtagname)
- [blocks::stream_to_tagged_stream](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-blocks/grc/blocks_stream_to_tagged_stream.block.yml#L47) (len_tag_key)
- [pdu::pdu_to_tagged_stream](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-pdu/grc/pdu_pdu_to_tagged_stream.block.yml#L33), [pdu::tagged_stream_to_pdu](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-pdu/grc/pdu_tagged_stream_to_pdu.block.yml#L34) (tag)
- [blocks::tag_gate](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-blocks/grc/blocks_tag_gate.block.yml#L55) (single_key)
- [network::tuntap_pdu](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-network/grc/network_tuntap_pdu.block.yml#L39) (ifn)
- [network::socket_pdu](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-network/grc/network_socket_pdu.block.yml#L49) (host, port)
- [all ZeroMQ blocks](https://github.com/gnuradio/gnuradio/tree/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/gr-zeromq/grc) - they use `const_cast<char*>(${address}${'.c_str())' if str(address)[0] not in '"\'' else ')'}`, which still does not compile without double quotes.

Perhaps, this can be fixed with the addition of `"'": '"'` translation, like the `sizeof` translation in [cpp_top_block.py](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb1c87eb53ee/grc/core/generator/cpp_top_block.py#L287-L289)?

marcusmueller, 2023-09-18:
He @AsriFox! Nice, thank you for the list :)

Problem really is that in some cases you need that (in case you're entering a string literal), in other cases, users want their raw input to be used verbatim in the source code…

Bit of historical background on that: back when GRC was invented (that was @guruofquality IIRC), there was no C++ generation, and the idea was that universally, the language that GRC was written in, the language these fields get interpreted within GRC, and the language of the generated code were all the same – Python. So, no special care was needed to support "cross-language entry".

I think the list you gave us is still quite excellent, because, well, these are things that can be fixed right now; this feels especially unnecessary:

Condition|Python Output|C++ Output
-------------|----------|----------
String parameter with no quotes|String in single quotes (valid)|String in single quotes **(invalid)**
String parameter is empty|String in single quotes (valid)|String in single quotes **(invalid)**

Do you maybe want to have a stab at these two entries in the table (and others, as you see fit, but my guess is these are the easiest to fix to begin with)?

AsriFox, 2023-09-23:
I have found that `r"'([^']*)'": r'"\1"'` translation can solve the issue for simple string parameters:
| Condition | GRC | Display | C++ Output |
|-|-|-|-|
| String parameter with no quotes | `host: 127.0.0.1` | 127.0.0.1 | `"127.0.0.1"` |
| String parameter in single quotes | `host: '''127.0.0.1'''` | 127.0.0.1 | `"127.0.0.1"` |
| String parameter is empty | `host: ''` | | `""` |
| String parameter is set to empty single-quotes | `host: ''''''` | | `""` |
| String parameter is set to empty double-quotes | `host: '""'` | | `""` |
| String parameter in double quotes | `host: '"127.0.0.1"'` | 127.0.0.1 | `"127.0.0.1"` |
| Raw string literals | `host: R'127.0.0.1'` | 127.0.0.1 | `R"127.0.0.1"` |

The proposed translation will work most of the time. Hovever, this pattern replaces **all** pairs of single quotes, including nested ones:
| Condition | GRC | Display | C++ Output |
|-|-|-|-|
| String parameter with nested single-quotes | `host: '''''127.0.0.1'''''` | ''127.0.0.1'' | `"""127.0.0.1"""` |
| Multi-part string with single-quotes | `host: '''127.0''''.0.1'''` | 127.0.0.1 | `"127.0"".0.1"` |
| Multi-part string with double-quotes | `host: '"127.0"".0.1"'` | 127.0.0.1 | `"127.0"".0.1"` |
| Single-quotes inside double-quotes | `host: '"127.0''''.0.1"'` | 127.0''.0.1 | `"127.0"".0.1"` |
| Double-quotes inside single-quotes | `host: '''127.0"".0.1'''` | 127.0"".0.1 | `"127.0"".0.1"` |

While the provided examples are somewhat artificial, Python can work with them. This confusion happens because the translation code substitutes templated Python code into C++ code without context. This is probably fine, as long as no blocks require complex strings (e.g. regex); in that case *dtype* can be changed to 'raw'.

Example flowgraph:
[cpp_string_bug.grc.txt](https://github.com/gnuradio/gnuradio/files/12707027/cpp_string_bug.grc.txt)

marcusmueller, 2024-01-03:
@AsriFox is on it

PR #6972 (2024-01-28): GRC: generate valid C++ string parameters without explicit double quotes
## Description
When generating C++ code, GRC did not have a standard way to translate Python strings to C++ strings, because Mako uses simple text substitution. To address this, additional translations are added to generators for blocks and parameters/variables with the first commit.

Translations already used in the block definitions were removed with the second commit:
- manual translations with `no_quotes` helper, like in [qtgui::waterfall_sink_x](https://github.com/gnuradio/gnuradio/blob/9b3183aec7b629fd7fbb2b1574f0bb [... cut]
closed 2024-01-28 by willcode