gnuradio/gnuradio #2522: argument names missing from docstrings

dforsi, 2019-06-02:
For example, the docstring for the file_source block should read

> Opens filename as a source [...]
> The itemsize of the block  [...]
> If repeat is turned on [...]

instead it reads

> Opens  as a source [...]
> The  of the block [...]
> If  is turned on [...]

(ie the words "filename", "itemsize" and "repeat" are missing).

Steps to reproduce:
````
from gnuradio import blocks
print(blocks.file_source.__doc__)

        make(size_t itemsize, char const * filename, bool repeat=False) -> file_source_sptr

        Read stream from file.

        Constructor Specific Documentation:

        Create a file source.

        Opens  as a source of items into a flowgraph. The data is expected to be in binary format, item after item. The  of the block determines the conversion from bits to items.

        If  is turned on, the file will repeat the file after it's reached the end.

        Args:
            itemsize : the size of each item in the file, in bytes
            filename : name of the file to source from
            repeat : repeat file from start
````
Argument names are missing also in the [Sphinx documentation](https://www.gnuradio.org/doc/sphinx/blocks_blocks.html#gnuradio.blocks.file_source) but they are visible in the [Doxygen documentation](https://www.gnuradio.org/doc/doxygen/classgr_1_1blocks_1_1file__source.html#a21a22625224ef17b78524aea323bab27).

tpiepho-synapse, 2020-01-22:
I think this is happening somewhere inside the doxyxml Python code and how it parses the XML documentation extracted from the doxygen comments.  For example, this in the xml code for threshold_ff docs:

```xml
<para>Test the incoming signal against a threshold. If the signal excedes the <computeroutput>hi</computeroutput> value, it will ...
```

When the doxyxml code gets this data, the detailed description of the class will be a list of `doxyxml.generated.compoundsuper.MixedContainer` objects, which will be:

```
'Test the incoming signal against a threshold. If the signal excedes the ',
'value, it will ...',
...
```
The bit in the `<computeroutput>` block gets dropped of the list of content items and thus the final textual version of the documentation.

tpiepho-synapse, 2020-01-23:
The issue appears to be with the generated python code that parses the xml.  The schema probably didn't include `<computeroutput>'.  Parsing the xml gets down to handling the children of an object of `compoundsuper.docParaType` and the child node of type `ELEMENT_NODE` with the computeroutput parameter name in it is just dropped because a `<para>` didn't include the possibility of having a `<computeroutput>` child in the schema.

Unfortunately, the generated python source is what is checked into git, rather than the schema and the build logic to use the schema to generate that source using generateDS.  So it's not easy to commit a fix.

marcusmueller, 2024-01-03:
Would be nice if `blocks.file_source.__doc__` actually contained block information, but the parameters with our Pybind11 build are correctly documented in `blocks.file_source.__init__.__doc__`.

closed 2024-01-03 by marcusmueller