gnuradio/gnuradio #2536: grc: Help --> About box displays nothing and gives error

mormj, 2019-06-06:
Clicking on the Help-->About menu item in grc has no effect - no dialog pops up, and in the terminal used to launch gnuradio-companion, the following error is displayed:

Traceback (most recent call last):
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/grc/gui/Dialogs.py", line 274, in show_about
    ad.set_logo(Gtk.IconTheme().load_icon('gnuradio-grc', 64, 0))
GLib.Error: gtk-icon-theme-error-quark: Icon 'gnuradio-grc' not present in theme (null) (0)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/grc/gui/Application.py", line 445, in _handle_action
    Dialogs.show_about(main, self.platform.config)
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/grc/gui/Dialogs.py", line 276, in show_about
    log.debug("Failed to set window logo")
NameError: name 'log' is not defined

haakov, 2019-06-21:
It works fine for me, but my master branch is a couple weeks old. (https://github.com/gnuradio/gnuradio/commit/73b074f121b0ab2ac38336916a60891c4d88d2cb is the latest commit)

bastibl, 2019-06-21:
Had the same problem. It seems to happen if the freedesktop stuff is not installed (in the build directory `bash ./grc/scripts/freedesktop/grc_setup_freedesktop install`) Anyway, I think the exception handler in Dialogs.py is broken since the logger is not created.

dl1ksv, 2019-06-23:
Got the same problems with the lastest commit. But the icons are in CMAKE_INSTALL_PREFIX/share/icons

But looking at 
Gtk.IconTheme().get_search_path()
this directory is not part of the search path.

As ~/.local/share/icons is part of the search path I copied the icons subdirectory to this position and it works.
So adding CMAKE_INSTALL_PREFIX/share/icons to the icon_search_path could solve this issue.
The name error has to be fixed separately.

dl1ksv, 2019-06-26:
An install script grc_setup_freedesktop is generated, but will not be executed during the installation process. So if CMAKE_INSTALL_PREFIX/share is not in the XDG_DATA_DIRS path  list the icons won't be found.
The easiest way to fix is to execute the generated script during installation. see #2580

mormj, 2019-07-18:
This appears to be fixed with #2579

closed 2019-07-18 by mormj