gnuradio/gnuradio #2546: make test (qa_volk_32f_log2_32f) fail

166MMX, 2019-06-11:
Apple LLVM version 10.0.0 (clang-1000.10.44.4)
Target: x86_64-apple-darwin17.7.0
Thread model: posix

```
$ git clone \
        --recurse-submodules \
        --depth 1 \
        --branch v3.7.13.5 \
        https://github.com/gnuradio/gnuradio.git
$ git describe 
v3.7.13.5
$ git status
Not currently on any branch.
nothing to commit, working tree clean
$ ctest -V -R qa_volk_32f_log2_32f
UpdateCTestConfiguration  from :/Users/build/jobs/gnuradio/builds/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/build/jobs/gnuradio/builds/DartConfiguration.tcl
Test project /Users/build/jobs/gnuradio/builds
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 38
    Start 38: qa_volk_32f_log2_32f

38: Test command: /bin/sh "/Users/build/jobs/gnuradio/builds/volk/lib/volk_32f_log2_32f_test.sh" "/Users/build/jobs/gnuradio/builds/volk/lib"
38: Test timeout computed to be: 10000000
38: RUN_VOLK_TESTS: volk_32f_log2_32f(131071,1)
38: generic completed in 1.203ms
38: a_sse4_1 completed in 0.24ms
38: u_generic completed in 1.502ms
38: u_sse4_1 completed in 0.192ms
38: offset 1310 in1: -1.71983e-07 in2: -8.10623e-06 tolerance was: 3
38: volk_32f_log2_32f: fail on arch a_sse4_1
38: offset 1310 in1: -1.71983e-07 in2: -8.10623e-06 tolerance was: 3
38: volk_32f_log2_32f: fail on arch u_sse4_1
38: Best aligned arch: u_sse4_1
38: Best unaligned arch: u_sse4_1
1/1 Test #38: qa_volk_32f_log2_32f .............***Failed    0.05 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.17 sec

The following tests FAILED:
	 38 - qa_volk_32f_log2_32f (Failed)
Errors while running CTest
```

michaelld, 2019-06-11:
Can you test the latest Volk GIT master? To the best of my knowledge, we've tweaked all of the QA to either work correctly, or just removed those that don't work any longer. All tests pass "make test" for my on multiple OSs now. I don't recall what the GR GIT component Volk commit is pegged to, but I'm guessing it's older than the current Volk GIT master.

166MMX, 2019-06-11:
Building agains current Volk GIT master let all test pass.
`100% tests passed, 0 tests failed out of 339`
The hash I build against:
```
$ git -C /Users/build/jobs/gnuradio/workspace submodule update --remote
Submodule path 'volk': checked out '10c0e5696433aae658d1fecc767267e5b9ae5e26'
```
```
$ ctest -V -R qa_volk_32f_log2_32f
UpdateCTestConfiguration  from :/Users/build/jobs/gnuradio/builds/8/DartConfiguration.tcl
UpdateCTestConfiguration  from :/Users/build/jobs/gnuradio/builds/8/DartConfiguration.tcl
Test project /Users/build/jobs/gnuradio/builds/8
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 38
    Start 38: qa_volk_32f_log2_32f

38: Test command: /bin/sh "/Users/build/jobs/gnuradio/builds/8/volk/lib/volk_32f_log2_32f_test.sh" "/Users/build/jobs/gnuradio/builds/8/volk/lib"
38: Test timeout computed to be: 10000000
38: RUN_VOLK_TESTS: volk_32f_log2_32f(131071,1)
38: generic completed in 2.326 ms
38: a_sse4_1 completed in 0.211 ms
38: u_generic completed in 1.465 ms
38: u_sse4_1 completed in 0.199 ms
38: Best aligned arch: u_sse4_1
38: Best unaligned arch: u_sse4_1
1/1 Test #38: qa_volk_32f_log2_32f .............   Passed    0.05 sec

The following tests passed:
	qa_volk_32f_log2_32f

100% tests passed, 0 tests failed out of 1

Total Test time (real) =   0.20 sec
```

michaelld, 2019-06-11:
OK; yeah that's what I thought would be the case. Thanks! I believe we're waiting to update the Volk submodule until we're closer to the actual GR 3.8.0.0 release. I also think we need a new Volk release too, which will happen before the GR release in order to get the submodule in place.

michaelld, 2019-06-11:
I'm going to go ahead and close this issue, since it's fixed in GIT and will eventually be in the Volk release && GR GIT submodule.

closed 2019-06-11 by michaelld