gnuradio/gnuradio #2547: Output of `make test` for failed tests is useless

rear1019, 2019-06-11:
Output of `make test` or `ctest --verbose` [1] for failed tests is useless. It does not tell which test exactly failed, e.g.:
```
$ ctest --verbose -R atsci
[…]
test 193
    Start 193: atsci-test

193: Test command: /usr/bin/sh "/home/sikon/dev/gnuradio/build/gr-atsc/lib/atsci-test_test.sh"
193: Test timeout computed to be: 10000000
193: ......F..............  Summary: 0 decoder errors out of 50 trials.
193: .  Summary: 0 decoder errors out of 50 trials.
193: .  Summary: 0 decoder errors out of 500 trials.
193: ...........
193: 
1/1 Test #193: atsci-test .......................***Failed    0.77 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.78 sec

The following tests FAILED:
	193 - atsci-test (Failed)
Errors while running CTest
```

Only by chance I discovered that a XML file with information about the failed test is created. This file is in a hidden folder deep down in the build directory, e.g. `build/gr-analog/lib/.unittests/` and `build/gr-analog/python/analog/.unittests/`. Such important output should be displayed and not discoved by chance.

[1] `make ctest` calls `ctest` without the verbose flag (which is not that helpful anyway).

michaelld, 2019-06-11:
Some tests' verbosity is more useful than others, but it all depends on the level of verbosity programmed into the test itself. For example "atsci-test" clearly doesn't provide much verbosity, and, yes, you can find stored files that _sometimes_ provide more info. But, not always.

michaelld, 2019-06-11:
If you want to help out, you could look into the tests that are bugging you and add specific verbosity && then submit a PR here for your changes.

closed 2019-06-11 by michaelld

rear1019, 2019-06-11:
I am not speaking about “debug” output generated by `printf()` explicitly contained in tests. I am speaking about output generated automatically by `ctest` or the unit test library (or whatever). The XML files I mentioned belong to this output (at least this is my impression). As I mentioned, I only know of their existence by chance.

The point is: A single test can contain more than a single sub-test. See for example `gnuradio/gr-atsc/lib/qa_atsci`. I deliberately have made one of the sub-tests fail by making a `CPPUNIT_ASSERT` fail. The output of `ctest` (see first post) does not indicate which of the sub-tests failed. It’s not possible to figure that out given the output. This information is buried deep down in the build directory inside a XML file, e.g. `build/gr-atsc/lib/.unittests/gr_atsc.xml`:
```xml
<?xml version="1.0" encoding='ISO-8859-1' standalone='yes' ?>
<TestRun>
  <FailedTests>
    <FailedTest id="6">
      <Name>qa_atsci_reed_solomon::t0_reed_solomon</Name>
      <FailureType>Assertion</FailureType>
      <Location>
        <File>/home/sikon/dev/gnuradio/gr-atsc/lib/qa_atsci_reed_solomon.cc</File>
        <Line>116</Line>
      </Location>
      <Message>assertion failed
- Expression: decoder_errors == 1
</Message>
    </FailedTest>
  </FailedTests>
<!--[…]-->
```

michaelld, 2019-06-11:
OK; thanks for clarifying. Yes, you are totally correct here. CMake provides immediate feedback on at most 2 levels of testing groups. "make test" provides only the very high level grouping, while "ctest -V" provides the 2nd level grouping. Neither provide the level of grouping you mention here. We've tried to move the majority of our testing into the 2nd level so that each will show that specific test's success or failure. That said, there are still some at the 3rd level of grouping, which means if any one of those fails then the whole test fails && it's not obvious from CMake which failed ... as you note without diving into the CMake generated XML ... which is in itself non-obvious how to track down and parse. I'd love to see more of our tests be at the 2nd level ... as noted: we'd love PRs to help tweak these issues to be more user friendly!