gnuradio/gnuradio #2554: Feature request: SI prefixes for numeric inputs

btlvr, 2019-06-15:
I think SI prefixes should be recognized by GNU Radio Companion. 

If I supply `2000` to a block parameter, this is displayed as `2k` on the canvas. It would be very convenient to be able to supply `2k`. At the very least, I think the notation should be consistent. If GRC displays a value with an SI prefix, it should be able to accept one as well.

aru31, 2019-06-18:
Hi @btlvr,
This feature already exists in GNU Radio Companion.
Pull request: [2352](https://github.com/gnuradio/gnuradio/pull/2352)

777arc, 2019-06-22:
@aru31 It doesn't seem to work for me.  If I make a new flowgraph, add a Constant Source and QT GUI Time Sink, connect them, run it, it works like normal.  Then I change the Constant Source value to "3k", and I get SyntaxError: invalid syntax:

self.analog_const_source_x_0 = analog.sig_source_c(0, analog.GR_CONST_WAVE, 0, 0, 3k)

Looks like the "3k" is still showing up.  Does that simple test work for you?  I'm on the latest master and I see the code change, but maybe my install is screwed up.

aru31, 2019-06-22:
@777arc: sorry, I couldn't find the block. Can you guide me the procedure to get that block?
Thanks. :)

777arc, 2019-06-22:
Well it doesn't really matter which block you use.  You can substitute the QT GUI Time Sink for any sink block.  Can you describe a minimal example that works, so I can test it out?

aru31, 2019-06-23:
@777arc: you were right, the flowgraph does generate an error. The above pull request was for issue #2322, which was for accepting the prefix for display and editing in the GRC. It doesn't solve the above problem as you mentioned, maybe because the variable inputs for generating a flowgraph are not the same which have been parsed to display the same in the GRC.

marcusmueller, 2024-01-03:
done

closed 2024-01-03 by marcusmueller