gnuradio/gnuradio #2570: Trouble with installation of GNU Radio from the source

haronix-bg, 2019-06-22:
Hi,

I have a trouble with installation of GNU Radio from the source on Ubuntu 18.04: Here is the part of the log:

-- c flags:  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall;
-- asm flags:  
-- c flags:  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized -Wall;
-- asm flags:   
-- Loading version 1.4.1git into constants...
-- Using install prefix: /usr/local
-- 
-- Configuring volk support...
--   Enabling volk support.
--   Override with -DENABLE_VOLK=ON/OFF
--   Override with -DENABLE_INTERNAL_VOLK=ON/OFF
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- 
-- Configuring doxygen support...
--   Dependency DOXYGEN_FOUND = YES
--   Enabling doxygen support.
--   Override with -DENABLE_DOXYGEN=ON/OFF
-- 
-- Configuring sphinx support...
--   Dependency SPHINX_FOUND = TRUE
--   Enabling sphinx support.
--   Override with -DENABLE_SPHINX=ON/OFF
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 

-- Configuring gr-ctrlport support...
--   Dependency Boost_FOUND = 1
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Enabling gr-ctrlport support.
--   Override with -DENABLE_GR_CTRLPORT=ON/OFF
-- Loading build date Sat, 22 Jun 2019 21:22:55 into constants...
-- Loading version 3.8tech-preview-418-g46995422 into constants...
-- Checking for module 'thrift'
--   No package 'thrift' found
-- Binary 'thrift' not found.
-- TRY_SHM_VMCIRCBUF set to ON.

-- Configuring gr-dtv support...
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_FEC = ON
--   Dependency ENABLE_VOLK = ON
--   Enabling gr-dtv support.
--   Override with -DENABLE_GR_DTV=ON/OFF
-- 
-- Configuring gr-audio support...
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Enabling gr-audio support.
--   Override with -DENABLE_GR_AUDIO=ON/OFF
-- Found ALSA 1.1.3
-- Checking for module 'jack'
--   No package 'jack' found
-- Checking for module 'portaudio-2.0'
--   No package 'portaudio-2.0' found

-- Configuring gr-channels support...
--   Dependency ENABLE_VOLK = ON
--   Dependency Boost_FOUND = 1
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_GR_BLOCKS = ON
--   Dependency ENABLE_GR_FFT = ON
--   Dependency ENABLE_GR_FILTER = ON
--   Dependency ENABLE_GR_ANALOG = ON
--   Enabling gr-channels support.
--   Override with -DENABLE_GR_CHANNELS=ON/OFF
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.6m.so (found suitable version "3.6.8", minimum required is "2") 
-- Python checking for PyQt5 - found
-- Checking for module 'Qt5Qwt6'
--   No package 'Qt5Qwt6' found
-- QWT Version: 6.1.3

It's generated via cmake -DENABLE_GRC=ON -DCMAKE_BUILD_TYPE=Debug -g -O2 -03 -DPYTHON_EXECUTABLE=/usr/bin/python3 ../

The downloaded source is from git -c http.sslVerify=false clone --recursive http://git.gnuradio.org/git/gnuradio.git (SSL is not updated).

Any idea what is wrong?

michaelld, 2019-06-22:
It would be more useful to have the full log than just this part of it ... for future reference ...

Looks like you don't have GMP or MPIR installed. You can choose which one, or both. Might be other issues but that's what stands out to me.

haronix-bg, 2019-06-22:
Hi, @michaelld,

GMP and MPIR was installed from the source, after that deinstalled and finally installed with sudo apt get install libgmp-dev & libzmq3-dev but with no success.

michaelld, 2019-06-22:
Can you clear out the build directory and redo the whole cmake command and pipe or tee the output to a file that you can then attach to this issue? Seeing the whole cmake output will really help here.

michaelld, 2019-06-22:
Check out the instructions here: https://wiki.gnuradio.org/index.php/UbuntuInstall#Bionic_Beaver_.2818.04.29

haronix-bg, 2019-06-23:
Yes, I saw that link 7 days ago, but after debugging few days the result is the same.
You can find the full log on the following link:

https://ufile.io/yr7ueekw

haronix-bg, 2019-06-23:
Other question: what is the recommended version of Linux distribution and other versions of compilers like gcc g+ and etc for Manual Source Build? Is it working the currently version of source code published in github?

michaelld, 2019-06-23:
Reviewing your cmake log, all that's missing is UHD. If you "make" this build, does it succeed?

michaelld, 2019-06-23:
Any compiler that supports c++11 should work. Most linux distros will also work, so long as they can install dependencies ... obviously of you can use "yum" or "apt" or whatever to get background dependencies that makes things easier!

balister, 2019-06-24:
Fedora uses dnf now, not yum.

On 06/23/2019 04:11 PM, Michael Dickens wrote:
> Any compiler that supports c++11 should work. Most linux distros will also work, so long as they can install dependencies ... obviously of you can use "yum" or "apt" or whatever to get background dependencies that makes things easier!
>

haronix-bg, 2019-06-29:
https://github.com/gnuradio/gnuradio/issues/2105 it has to be open, the trouble is still available :D

closed 2024-03-11 by marcusmueller