gnuradio/gnuradio #2583: QT Windows flickering

mberendsen, 2019-06-28:
GNU Radio Companion 3.7.13.5
OSX 10.14.1
XQuarts 2.17.11

I am using a simple GRC to pass hackrf data from osmocom Source to QT Freq sink. When the XQuarts window is created by python it begins flickering. It looks like it flickers with each redrawl of the window but I'm not sure. the rate of flickering seems affected by sample rate. The QT graph/objects are not affected by the flickering only the generated XQuarts window.

This continues happening when moving between XQuarts windows but when I activate another application window like chrome the flickering stops and does not start even when the XQuarts windows are activated again.

fmagno, 2019-07-04:
I think I had this issue before. Try changing the `style` setting in file `~/.gnuradio/config.conf` from `raster` to `native` as follows:

```
(...)
[qtgui]
#style = raster
style = native
```

ra1nb0w, 2019-10-01:
thank you so much @fmagno !

michaelld, 2019-10-01:
Yes what @fmagno writes is the correct solution to get GR37's Qt4 GUI working on modern OSX. I'm closing this issue, since there's no better way to do this.

closed 2019-10-01 by michaelld

brodietwallace, 2020-02-10:
As a followup to this previously closed issue, this fix no longer appears to work. 

My operating specifications:
GNU Radio Companion 3.7.13.5 installed from macports
OSX 10.14.5
XQuarts 2.17.11

Within gnuradio/config.conf I have:
[qtgui]
style = native

Additionally I have tried using the opengl setting (as reccomended here http://ettus.80997.x6.nabble.com/USRP-users-macOS-Mojave-10-14-and-GNU-Radio-UHD-Volk-td11083.html):
[qtgui]
style = opengl

Neither of these solutions succeeded, nor did placing the config.conf file into gnuradio/conf.d

michaelld, 2020-02-10:
Just to be clear, the file your using is  `~/.gnuradio/config.conf` ...
yes? I have the exact same setup as you & tweaking this setting changes the
whole Qt GUI behavior for me. `raster` flashes; `native` looks great but
can be slow; `opengl` is faster but some things don't show up correctly
(e.g., sliders). Qt4 is getting old whole macOS continues to move forward
... it's really no great surprise that we're having display issues with it!
- MLD