gnuradio/gnuradio #2584: Variable with a big list makes GRC UI unresponsive/slow

fmagno, 2019-06-28:
The following flowgraph comprised of a single variable `big_list` with a large number of items, e.g 1000000, makes the UI unresponsive or very slow.

[image]

fmagno, 2019-06-29:
The issue is also visible @ `v3.8-MacPorts-next-git-8a7214c7(20190521-8a7214c7) (Python 2.7.16)`

marcusmueller, 2024-01-03:
persists, since the evaluation happens all over the namespace, and ends up in the variable explorer, which has the most beautiful buffer addressing overflows, it seems:

[image]

[image]

marcusmueller, 2024-03-11:
categorizing "low" because of the existence of the Qt GRC