gnuradio/gnuradio #2585: GRC does not show variable names

fmagno, 2019-06-29:
`GR version: v3.8-MacPorts-next-git-8a7214c7(20190521-8a7214c7) (Python 2.7.16)`

gnuradio-companion does not show variable name inside the variable block. It can be confusing when dealing with flowgraphs with many variables.

[image]

bastibl, 2019-07-01:
Can you please update GNU Radio and try again (if I understand the build description correctly, you are using the GNU Radio version from 14 May).  There was an issue about the problem (https://github.com/gnuradio/gnuradio/issues/2235) and it should be fixed in the most recent version (https://github.com/gnuradio/gnuradio/pull/2568).

fmagno, 2019-07-01:
OK, thanks. I'll give it a try as soon as I can. For now it seems wise just to close this issue.

closed 2019-07-01 by fmagno