gnuradio/gnuradio #2590: GR 3.8 | top_block.wait() does not flush/close open file

fmagno, 2019-07-02:
I believe that `wait()` should not return until all files have been flushed/closed. Do you guys agree?

Follows an example:

[image]

```python
# my_app.py

from foo2 import foo2

f = foo2()
f.run()
f.wait()
# foo2.dat is an empty file at this point
del f    # flush occurs now
# foo2.dat has now the expected content
```

bastibl, 2019-07-02:
AFAIS, the underlying problem is that the file sink does not implement `stop()` to close the file.

mormj, 2019-07-02:
I can't repeat this - from both 3.7 and 3.8 I get a 3 byte file with the automatically generated flowgraph top_block.py

which just has this: 
```
def main(top_block_cls=top_block, options=None):
    tb = top_block_cls()
    tb.start()
    tb.wait()
```

[image]

mormj, 2019-07-02:
Ooooh I get it now.  In my case, the file is only written when the flowgraph exits, but if I put a sleep after the wait, there is an empty file until the flowgraph exits.  Not sure what the desired behavior should be in this case.

mormj, 2019-07-02:
And same behavior on 3.7 - probably can remove the 3.8 from the title

fmagno, 2019-07-03:
Hi @mormj, IMO as soon as `wait`, `run` or `stop` return all tasks related to the execution of the fg should be finished, including flushing and closing files.

mormj, 2019-07-08:
@fmagno - I agree - that is what I would expect the behavior to be

rear1019, 2020-03-29:
Mh, this is not a matter of simply calling ``file_sink_base::close()``. ``file_sink_base`` (base class for ``file_sink_impl``) maintains two file descriptors and a state variable. I am afraid to touch any of them given that I don’t understand the code completely. 

PS: ``wavfile_sink_impl`` contains a copy of ``file_sink_base``’s code.

PR #3357 (2020-04-13): file_sink: Flush file on stop()
Make sure that data passed to the file sink is written to the disk when
the flowgraph is stopped. Before this change this only happened
implicitly due to a fclose() in the base class destructor. This in turn
means that data was written delayed after stopping the flowgraph [1].

This fixes #2590.

[1] https://github.com/gnuradio/gnuradio/issues/2590
commit ec8a9284 (2020-04-13): file_sink: Flush file on stop()

Make sure that data passed to the file sink is written to the disk when
the flowgraph is stopped. Before this change this only happened
implicitly due to a fclose() in the base class destructor. This in turn
means that data was written delayed after stopping the flowgraph [1].

This fixes #2590.

[1] https://github.com/gnuradio/gnuradio/issues/2590
commit 1ba373bb (2020-04-15): file_sink: Flush file on stop()

Make sure that data passed to the file sink is written to the disk when
the flowgraph is stopped. Before this change this only happened
implicitly due to a fclose() in the base class destructor. This in turn
means that data was written delayed after stopping the flowgraph [1].

This fixes #2590.

[1] https://github.com/gnuradio/gnuradio/issues/2590
commit 0d56c2a8 (2020-04-15): file_sink: Flush file on stop()

Make sure that data passed to the file sink is written to the disk when
the flowgraph is stopped. Before this change this only happened
implicitly due to a fclose() in the base class destructor. This in turn
means that data was written delayed after stopping the flowgraph [1].

This fixes #2590.

[1] https://github.com/gnuradio/gnuradio/issues/2590
closed 2020-04-13 by marcusmueller