gnuradio/gnuradio #2597: Complex to IChar and IShort Saturation and Data Loss

ghostop14, 2019-07-08:
Hi folks, I've been hunting down this conversion issue for a while with streaming SC8 data and I think I can finally articulate the issue I'm seeing.  The problem exists for both Complex<->IChar and Complex<->IShort conversions.  Both of these do have a note in their work functions that the developer was aware converting to the smaller data type could "saturate" it.  But it may never have been addressed.

So the problem is if you take FC32 input from say any SDR source block (I was using an Airspy in an OSMOSDR source block), then go Complex->IChar and immediately IChar->Complex, the resulting output does not match the input.  Further investigation into the CSDR tool's successful conversions to/from SC8<->FC32 and the IShort versions, as well as how the OSMOSDR source block is converting the underlying 8-bit and 12-bit samples to floats for block output show a scaling factor such that the correct conversions should look like:
Complex->Signed 8
*out++ = (signed char)lrintf(*in++ * (float)SCHAR_MAX)
Then Signed 8 to complex like this:
*out++ = ((float)(*in++))/(float)SCHAR_MAX

This seems to prevent the clipping better with the scaling factor rather than just lrintf(*in++) that's in the current IChar/IShort blocks.  Testing with the above code showed successful / matching results when going Source->[Complex->SC8]->[SC8->Complex]->Freq Sink using these calculations versus the current IChar versions.

If you simply do a conversion from a source like USRP->[IShort to Complex] you won't run into this issue.  It's only when you try to symetrically go in/out the clipping seems to occur.  Which is a problem if you're trying to reduce the data size by storage or downconverting to a smaller byte format for network transmission then back for processing on a receiving end.

Note: The same issue holds true for the IShort conversions using SHRT_MAX as the scaling factor.

Thanks!

nlhnt, 2020-12-16:
Thank you for your investigation!
I am not part of the GNU Radio team but some screenshots with examples of this happening would have been great!

rear1019, 2020-12-27:
Closing because better documentation a scale parameter have been added in pull request #3208.

Strictly speaking the scale parameter is not required as one could simply add a multiply block beforehand. On the other hand it does make sense performance-wise.

@ghostop14 Please reopen if this bug report is still relevant

closed 2020-12-27 by rear1019