gnuradio/gnuradio #2603: Module 'Bluefile' has no attribute 'bfSource'

pianoman730, 2019-07-09:
After much searching and debugging on multiple fronts, I'm creating an issue here to try and see if this is a bug or not. **TLDR;** There appears that there is an issue in how swig links between the c++ code and the python code in an OOT block.

I'm using [this](https://github.com/gnuradio/gnuradio/tree/27dd99e49a7d34eeb4c0a8fedfe3b072ad582662) commit as the base for GR, as that is what `master` was at the time I installed it. I built from source on Centos 7 with Python 3.6.

**Details:**
I have created an OOT module called 'Bluefile'. This module currently has two blocks that I generated with `gr_modtool`: `bfSource` and `bfSink`. It is basically trying to process custom files. With the code that I have currently, it compiles and installs just fine, but when I place the block on the canvas in GRC and run it, I get this error:
```
Generating: '/home/<user>/Desktop/top_block.py'

Executing: /bin/python3 -u /home/<user>/Desktop/top_block.py

Traceback (most recent call last):
  File "/home/<user>/Desktop/top_block.py", line 162, in <module>
    main()
  File "/home/<user>/Desktop/top_block.py", line 150, in main
    tb = top_block_cls()
  File "/home/<user>/Desktop/top_block.py", line 118, in __init__
    self.Bluefile_bfSource_0 = Bluefile.bfSource('/home/<user>/Downloads/whitenoise.txt', 1000, CI, 32768, 2)
AttributeError: module 'Bluefile' has no attribute 'bfSource'

>>> Done (return code 1)
```
I have done a couple of things to track this error down, and I'm concluding with the work that I've done on it that there is an issue in how swig is generating the files. Here's why:

After installing the block with `sudo make install` from my build directory, I went into the code to try track back the error. From the `top_block.py` file, I opened the file that `Bluefile.bfSource` comes from. This file is `/usr/local/lib/python3.6/dist-packages/Bluefile/__init__.py`. This file contains this block:
```
try:
    # this might fail if the module is python-only
    from .Bluefile_swig import *
except ImportError:
    pass
```
I changed the except block to:
```
except ImportError as ex:
    print(ex)
    pass
```
From the command line I run `python3` and run `import Bluefile` and `No module named '_Bluefile_swig'` is printed. This led me to look at how `Bluefile_swig.py` is being imported. The top of that file has this:
```
from sys import version_info as _swig_python_version_info
if _swig_python_version_info >= (2, 7, 0):
    def swig_import_helper():
        import importlib
        pkg = __name__.rpartition('.')[0]
        mname = '.'.join((pkg, '_Bluefile_swig')).lstrip('.')
        try:
            return importlib.import_module(mname)
        except ImportError:
            return importlib.import_module('_Bluefile_swig')
    _Bluefile_swig = swig_import_helper()
    del swig_import_helper
elif _swig_python_version_info >= (2, 6, 0):
    def swig_import_helper():
        from os.path import dirname
        import imp
        fp = None
        try:
            fp, pathname, description = imp.find_module('_Bluefile_swig', [dirname(__file__)])
        except ImportError:
            import _Bluefile_swig
            return _Bluefile_swig
        try:
            _mod = imp.load_module('_Bluefile_swig', fp, pathname, description)
        finally:
            if fp is not None:
                fp.close()
        return _mod
    _Bluefile_swig = swig_import_helper()
    del swig_import_helper
else:
    import _Bluefile_swig
del _swig_python_version_info
```
I'm using python 3.6 so this line `mname = '.'.join((pkg, '_Bluefile_swig')).lstrip('.')` is part of the culprit. I then changed `_Bluefile_swig` to `Bluefile_swig` and ran `python3` from the command line and importing `Bluefile` I get this error:
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/Bluefile/__init__.py", line 30, in <module>
    from .Bluefile_swig import *
  File "/usr/local/lib/python3.6/dist-packages/Bluefile/Bluefile_swig.py", line 115, in <module>
    class bfSource(object):
  File "/usr/local/lib/python3.6/dist-packages/Bluefile/Bluefile_swig.py", line 161, in bfSource
    __swig_destroy__ = _Bluefile_swig.delete_bfSource
AttributeError: module 'Bluefile.Bluefile_swig' has no attribute 'delete_bfSource'
```
This leads me to believe that there is an issue in how swig is linking. When changing that name to take out the underscore, the import is successful but then fails when trying to find another attribute in the file. 

Other things I've tried was linking directly to the generated .so file in that python installation directory via cmake and ldconfig. I've also tried to import `Bluefile_swig` directly and I get a segmentation fault. I also looked at the generated .so's of some of the built in core blocks and compared the attributes the the attributes in my OOT .so and found similar attributes as the ones that are NOT being found by python in my .so. So...yeah. Somethings wrong.

marcusmueller, 2019-07-15:
please put your whole OOT in a git repo somewhere, so that we can reproduce locally.

marcusmueller, 2019-07-15:
the thing is that the underscore *should* be there. It's the difference between the swig-generated .so wrapper `_oot_swig`, and the python module `oot_swig`

noc0lour, 2019-07-17:
@pianoman730 most likely there is an issue with the CMake setup in your OOT. It's possible this is a bug in the way how gr_modtool creates the CMake, in that case it needs fixing on our side. But without some details this is really hard to debug from our side. (I've successfully built OOTs for current GNU Radio master)' https://github.com/noc0lour/gr-grnet/tree/3.8 is what I usually refer to.

jamesrwills, 2020-04-29:
Hey, Jake. Did you ever get these Bluefile module working? I could really use bfSink and bfSource blocks. But, I don't see them in the repo.

JakeHadley, 2020-04-29:
This was a custom in house module that I created. So it's not shared open source, unfortunately. It was specifically for a code base that we were using.

jacobgilbert-snl, 2020-04-29:
https://github.com/sandialabs/gr-sandia_utils may help you. the file sink and source have limited bluefile support (provided you have the libraries, which we cannot distribute....but it sounds like you do)

A 3.8 update is coming very soon...

marcusmueller, 2024-01-03:
No source, no reproducibility, not GNU Radio's problem :)

closed 2024-01-03 by marcusmueller