gnuradio/gnuradio #2604: FFT: no shift possible for real input

Notou, 2019-07-10:
Hi everyone,

I was looking at the FFT block for documentation to the wiki, and I noticed that there is no way to specify a fftshift when the input is set to real and not complex. 
The [c++ impementation](https://github.com/gnuradio/gnuradio/blob/master/gr-fft/lib/fft_vfc_fftw.cc) does not handle it.

I'm not able to figure out why it's the case so my guess is that it's simply not yet implemented, but maybe I'm missing something.

777arc, 2019-09-16:
So, when the input is real, the output of the FFT will have the same values for the negative and positive portion.  Typically when I'm viewing the FFT of a real signal (outside of gnuradio) I'll only plot the positive portion.  So I'll essentially just throw out the 2nd half of the output vector, which is the negative frequencies when you haven't done an fftshift yet, and there's no need to fftshift because DC is already at index 0.

In general though I think people don't use the FFT block directly if they are trying to view the power spectral density, they will use the Frequency Sink.

michaelld, 2019-09-17:
Well explained. I think this is not really a big per se; it's a feature in this case. I'm closing this issue. If you disagree I'd suggest sending an email to the GR discussion list.

closed 2019-09-17 by michaelld