gnuradio/gnuradio #2607: Moving average, discontinuities in output

Notou, 2019-07-11:
Hi everyone,

There seem to be something wrong with the output of the Moving average block.
When I use it after a sine source (or other non constant source actually), there is a jump in the output values at the frequency of the max Iter value. Here is a flowgraph to reproduce it

[image]

And here is the output:
[image]

The scale of the jumps is quite small so it might get lost when there is some noise, but here it's quite noticeable.
These jumps also cause the average of the output (in this case, should be 0) to go up and down.

I'm using the 3.7.13.4 release, but I haven't found changes to the block work function since, so it should still be present in current master.

noc0lour, 2019-07-11:
I'd bet on underlying VOLK issues (we had kernels which had differing outputs for the same input if run on different architectures).
I currently don't know if VOLK exports old-style CMake variables to use with 3.7.13.4, I recall someone added them back in that case you could try to upgrade volk to master & rebuild your GNU Radio if you built from source.

marcusmueller, 2019-07-11:
@noc0lour  I haven't looked at MA in a while, but I think it's non-volkized but just keeps a running sum and a history and just subtracts the "oldest" sample and adds the newest one.

Notou, 2019-07-11:
Indeed, there is no volk involved.
It looks like something is not reset properly between calls to the work function but I didn't find what.

argilo, 2019-08-05:
I don't think there's any problem here. Your flowgraph is averaging 10 cycles of sine wave, which should always equal zero. What you're seeing in the time sink is just the floating point error that has accumulated over the summation.

argilo, 2019-08-05:
The discontinuities are visible because the block resets itself every maxiter samples to stop error from accumulating indefinitely.

Notou, 2019-08-06:
This is what I thought initially too. 
The image included  was not really telling so I got a new one:

[image]

Here the setup is the same with just the max iter parameter changed to 1000.
I agree it's perfectly normal to have a sin type shape with some variations due to some floating point errors.
But the jumps every max iter should make the output go back to having an average of 0, at least for the first couple of periods, right? And if you look after the second and fifth jumps, it's definitely not the case.

By the way, I can now say that the issue is present in 3.8 rc2.

bastibl, 2019-08-06:
I think what you see is not only the the error of the moving average block, but also the error introduced by the fixed-point sin/cos implementation used in the signal source. If you replace the signal source with a vector source (`np.sin(np.linspace(0,2*np.pi, 100))`) it looks as I would expect.

[image]

[image]

Notou, 2019-08-06:
Is it the sin like pattern you're talking about? If it is, I think I understand the issue for that. 
But there are still jumps once in a while and the average should still be 0, right?

bastibl, 2019-08-06:
I don't get your reply but to rephrase what I was trying to say: I think that
- the error you are seeing (and that you accredited to the moving average block) is actually the error of the fixed-point sin/cos implementation used in the signal source
- the fact that there is a small error even with a more precise sin/cos implementation is a result of the efficient implementation used the moving average block. If this is not accurate enough, you could use a FIR filter with all taps set to 1.

Notou, 2019-08-06:
If I take your image, I see 3 components of the error of the blue curve: 
- a sinus variation a the frequency of the input sinus,  <- What you think comes from the source?
- a progressive variation of the offset of that sinus of about 10% of the amplitude,
- sudden jumps of that offset up to more than 200% of the amplitude of the output sinus.

And I see one component in the green line: A sudden jump once in a while.

I understand that a progressively building up error is to be expected due to the moving window algorithm used for efficiency. 
But shouldn't it cause a situation where the output is almost 0, then slowly diverges (with some noise) for max iter samples where there is a jump back to 0, new divergence, jump to 0, ...?
If I'm not mistaken, it wouldn't be exactly 0 but the small amount of error coming from the addition on the Length first items. And that should be the same error as the one coming from the FIR filter, right?
But it's actually not the case. In this image, the blue line is from the moving average block after a signal source, with the mentioned issues, the red line is after a FIR filter with 1000 taps set to one and input from the vector source and the same parameters as you, and the green is after the moving average, input from the same vector source.
I expected the green line to match the red after each jump, but: 
[image]

I'm actually not trying to use the blocks for anything productive, I just saw something weird and I'm trying to understand what's going on.

bastibl, 2019-08-06:
> If I'm not mistaken, it wouldn't be exactly 0 but the small amount of error coming from the addition on the Length first items.

I don't know what you mean with "Length first items", but this is floating point arithmetic. So no, it won't be exactly 0 but something "close" to 0.

>And that should be the same error as the one coming from the FIR filter, right?

If you didn't change the parameters, you are comparing apples and oranges. In your screenshot you scale the Moving Average by 100. So you sum up 1000 samples and multiply the results with 100. If you want a one-to-one comparison with a FIR block, you'll have to use `[100]*1000` as taps.

If you do this, you see that the FIR filter and the Moving Average show (1) similar amplitudes and (2) the same *progressive variation*. So this is not a problem of the Moving Average block.

Regarding the jumps: I think, this just some floating point weirdness. Depending on how you do a calculation (order of summands, assembler instructions), it might show slightly different results. Try comparing a Moving Average with Max Iter=1 with a Moving Average with Max Iter=100000 block. You'll see that the Max Iter=1 block looks rather noisy and the Max Iter=100000 block jumps to value of the Max Iter=1 block after it reached its Max Iterations. So this looks good.

Furthermore, while calculating the sum, it reaches values of up to ~31 (`np.sum(np.sin(np.linspace(0,np.pi, 50)))`). When you set the scaling to 1, you'll see that the aggregated error of the fixed-point signal source and floating point errors after summing 1000 samples is below 0.0001. This is a relative error of about 3.2e-6 (i.e., 0.00032%). I'm not an expert about floating point arithmetic, but this doesn't seem unreasonable to me for a 32-bit float.

Notou, 2019-08-07:
> I don't know what you mean with "Length first items", but this is floating point arithmetic. So no, it won't be exactly 0 but something "close" to 0.

I meant the number of items corresponding to the parameter named Length, in this case, 1000. So yes, the amount of error introduced for computing the first value.

> If you didn't change the parameters, you are comparing apples and oranges. In your screenshot you scale the Moving Average by 100. So you sum up 1000 samples and multiply the results with 100. If you want a one-to-one comparison with a FIR block, you'll have to use [100]*1000 as taps.
Ahem, you're right, I forgot about that factor.

Lines with max iter != 1 do indeed match with max iter=1 after each jump and stay constant between jumps, possibly because the error introduced by the moving window algorithm is really small compared to the one from the initialization part.

You convinced me it's indeed not a bug but normal behaviour.
Thank you for taking the time to answer all my interrogations.

closed 2019-08-07 by Notou