gnuradio/gnuradio #2610: grc: block search applies hotkeys to flowgraph

mormj, 2019-07-11:
If a block is selected on the flowgraph, then typing a search in the block tree that includes the letters {'e','d','b', etc} applies those hotkeys to the flowgraph

E.g. if I search for 'digital' entering the 'd' will disable the block that is selected, then 'igital' in the search bar

drmpeg, 2019-07-11:
How do you get focus on both the block and the search bar at the same time? I can't do that on Linux/Ubuntu.

mormj, 2019-07-12:
This is how it looks from me pulling the latest upstream master @ cff05c36dfc3908c3:

[image]

Variable block is selected, went over to type "digital" in search box:

drmpeg, 2019-07-12:
Okay. I have to try it on Gnome then. I was SSH'ing to an 18.04 box from a 16.04 box with Unity.

mormj, 2019-07-12:
That's exactly the setup I was using too (16.04 ssh'd into 18.04).  I'll try on a different machine as well

drmpeg, 2019-07-12:
I was able to reproduce it. I think it only happens if you have "sloppy" focus enabled. You have to deselect the main window (by moving the mouse pointer off the window) with a block selected and then return the mouse pointer to the main window and click on the search bar.

haakov, 2019-07-13:
An earlier version of this bug: https://github.com/gnuradio/gnuradio/issues/2551

mormj, 2019-07-19:
This one still bothers me, but it looks like the underlying issue is that the hotkeys are global to the application, so when e,d,or b get pressed, the key press never makes its way into the search window.  

This could probably be solved by something kludgey such as in the ENABLE/DISABLE/BYPASS action handler, if the btwin has the focus, put the key in the search window and trigger the changed event

A cleaner solution is probably to make the e,d,b hotkey actions local only to the canvas??

PR #3710 (2020-08-19): grc: prevent search keystrokes from modifying flowgraph 
Fixes #2610.
Fixes #3507.

A confluence of bugs introduced while updating GRC to Python 3 results in keystrokes being sent to the drawing area while typing in the block search box.

First, the drawing area needs to be placed in a Viewport before being added into a ScrolledWindow. Fixing this allows the drawing area to grab focus as intended when it is clicked.

Second, the drawing area needs to unselect all blocks when it loses focus so that the relevant keystroke actions can be disabled. This functionality was commented out during the t [... cut]
closed 2020-08-19 by mbr0wn