gnuradio/gnuradio #2612: gr-uhd/3.8: RX channel .... out of range for configured RX frontends

devnulling, 2019-07-11:
## Issue Description ##
Using two channels with the UHD USRP Source block will throw an error
```
RX channel ... out of range for configured RX frontends
```
[image]

## Setup Details ##
- UHD 3.14.1.0
- GR master 3.8 @ https://github.com/gnuradio/gnuradio/commit/cff05c36dfc3908c3933773ac789959050fae45e
- USRP B210

## Expected Behavior ##
- No error

## Actual Behavior ##

```
Generating: '/home/user/top_block.py'

Executing: /usr/bin/python3 -u /home/user/top_block.py

[0;32m[INFO] [UHD] [0;39mlinux; GNU C++ version 7.3.0; Boost_106501; UHD_3.14.1.HEAD-0-gbfb9c1c7
[0;31m[ERROR] [UHD] [0;39mDevice discovery error: send_to: Operation not permitted
[0;31m[ERROR] [USRP2] [0;39mUSRP2 Network discovery error send_to: Operation not permitted
[0;31m[ERROR] [X300] [0;39mX300 Network discovery error send_to: Operation not permitted
[0;31m[ERROR] [UHD] [0;39mDevice discovery error: send_to: Operation not permitted
[0;32m[INFO] [B200] [0;39mDetected Device: B210
[0;32m[INFO] [B200] [0;39mOperating over USB 3.
[0;32m[INFO] [B200] [0;39mDetecting internal GPSDO.... 
[0;32m[INFO] [GPS] [0;39mFound an internal GPSDO: LC_XO, Firmware Rev 0.929a
[0;32m[INFO] [B200] [0;39mInitialize CODEC control...
[0;32m[INFO] [B200] [0;39mInitialize Radio control...
[0;32m[INFO] [B200] [0;39mPerforming register loopback test... 
[0;32m[INFO] [B200] [0;39mRegister loopback test passed
[0;32m[INFO] [B200] [0;39mPerforming register loopback test... 
[0;32m[INFO] [B200] [0;39mRegister loopback test passed
[0;32m[INFO] [B200] [0;39mSetting master clock rate selection to 'automatic'.
[0;32m[INFO] [B200] [0;39mAsking for clock rate 16.000000 MHz... 
[0;32m[INFO] [B200] [0;39mActually got clock rate 16.000000 MHz.
Traceback (most recent call last):
  File "/home/user/top_block.py", line 152, in <module>
    main()
  File "/home/user/top_block.py", line 130, in main
    tb = top_block_cls()
  File "/home/user/top_block.py", line 90, in __init__
    self.uhd_usrp_source_0.set_center_freq(98e6, 1)
  File "/home/user/sdr/installs/v3.14.1.0_master/lib/python3.6/dist-packages/gnuradio/uhd/uhd_swig.py", line 3599, in set_center_freq
    return _uhd_swig.usrp_source_sptr_set_center_freq(self, *args)
RuntimeError: LookupError: IndexError: multi_usrp: RX channel 35015184 out of range for configured RX frontends

>>> Done (return code 1)
```

## Steps to reproduce the problem ##
- Run attached flowgraph

## Additional Information ##
[b210_2ch.zip](https://github.com/gnuradio/gnuradio/files/3384252/b210_2ch.zip)

mwk088, 2019-09-30:
Has this been fixed yet?  Seeing that it is still open, I am guessing there is no updated release addressing this issue.

mormj, 2019-09-30:
Try changing the "Stream Channels" parameter from "[]" to "" (empty)

I believe #2823 will fix this issue once it is merged, but the above should work for now as a workaround.

mwk088, 2019-09-30:
Thanks.  That works.  I must have missed that comment somewhere.

Mark

closed 2019-11-01 by marcusmueller

nmhd1990, 2019-11-09:
I just want to use **baudline sink** in **gnuradio** as I have used it before by using **gr-baz**, but I do not know what is the problem exactly. I faced with the below problem. I am forced by the time, if someone could help me i appretiate that, thank you.

> Generating: '/home/me/top_block.py'
>>> Warning: This flow graph contains a throttle block and another rate limiting block, e.g. a hardware source or sink. This is usually undesired. Consider removing the throttle block.

Executing: /usr/bin/python -u /home/me/top_block.py

linux; GNU C++ version 5.3.1 20151219; Boost_105800; UHD_003.009.002-0-unknown

gr-osmosdr 0.1.4 (0.1.4) gnuradio 3.7.9
built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy redpitaya 
Using HackRF One with firmware 2014.08.1 
Using Volk machine: sse4_2_32_orc
['baudline', '-stdin', '-record', '-spacebar', 'recordpause', '-samplerate', '10000000', '-channels', '2', '-format', 'le32f', '-quadrature', '-basefrequency', '40900000.0']
[Errno 2] No such file or directory
Traceback (most recent call last):
  File "/home/me/top_block.py", line 141, in <module>
    main()
  File "/home/me/top_block.py", line 135, in main
    tb = top_block_cls()
  File "/home/me/top_block.py", line 107, in __init__
    **{})
  File "/usr/local/lib/python2.7/dist-packages/baz/baudline.py", line 146, in __init__
    self.p = subprocess.Popen(baudline_exec, stdin=subprocess.PIPE)	# , stdout=subprocess.PIPE, stderr=subprocess.PIPE, bufsize=16384 or -1
  File "/usr/lib/python2.7/subprocess.py", line 711, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1343, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

>>> Done (return code 1)

>

michaelld, 2019-11-09:
@nmhd1990 I can't see what your post has to do with this issue. This issue is closed because a PR was merged. Please open a new issue; or, even better, send an email to the GR discussion list "discuss-gnuradio@gnu.org"

PR #2823 (2019-11-01): gr-uhd: multichannel uhd object not populating channels
Due to the way the template interprets the stream_chans value as string,
it always returns True and the '[]' from stream chans was used as the
channels parameter.

With this change, channels in the case where stream chans is empty,
chans populates as before, which is range(num_chans)

Fixes #2789
Fixes #2612