gnuradio/gnuradio #2613: Document C++ code gen caveats

devnulling, 2019-07-11:
Testing out the new C++ code gen feature on master, ran into a few caveats that should be documented:

- The `Options` block ID must be changed to something other than `top_block`
- Within the `CMake options` field, CMake parameters must be separated by semicolons. 

For example:

`CMAKE_INSTALL_PREFIX=/path/to/prefix;UHD_DIR=/path/to/lib/cmake/uhd/;UHD_INCLUDE_DIRS=/path/to/include/;`

haakov, 2019-07-13:
Can be assigned to me

marcusmueller, 2019-07-15:
@haakov done assigning to you.

haakov, 2019-07-20:
About the CMake options, I'd like to drop the semicolons and just do:
`-DCMAKE_INSTALL_PREFIX=/path/to/prefix - DUHD_DIR=/path/to/lib/cmake/uhd/ - DUHD_INCLUDE_DIRS=/path/to/include/`
Which seems like the natural way to do it. I can't understand why I thought semicolons would be a good idea in the first place.

marcusmueller, 2024-01-03:
through multiple iterations of how to handle the id

closed 2024-01-03 by marcusmueller