gnuradio/gnuradio #2619: grc: adding default header variable produces a lot errors

dl1ksv, 2019-07-13:
Adding the default header variable object to a flowgraph produces the following errors:

ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate variable block variable_header_format_default_0
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/core/FlowGraph.py", line 226, in renew_namespace
    value = eval(variable_block.value, namespace, variable_block.namespace)
  File "<string>", line 1, in <module>
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/digital/digital_swig2.py", line 1677, in make
    return _digital_swig2.header_format_default_make(access_code, threshold, bps)
ValueError: invalid null reference in method 'header_format_default_make', argument 1 of type 'std::string const &'
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/gui/VariableEditor.py", line 195, in set_value
    evaluated = str(block.params['value'].evaluate())
KeyError: 'value'
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/gui/VariableEditor.py", line 195, in set_value
    evaluated = str(block.params['value'].evaluate())
KeyError: 'value'

The KeyError is correct as this variable contains no parameter named value !

As far as I can see all other variable block have a parameter named value that is used in the value assignmend of the block.

**Question:** Must all variable blocks must have a parameter named value ?

Despite the error meassage the flowgraph can be generated and the generated code seems to be correct

haakov, 2019-07-26:
Is this issue fixed?

dl1ksv, 2019-07-26:
Yes, this issue is fixed with #2662 
But #2605 is still open and should be applied.

closed 2019-07-26 by dl1ksv