gnuradio/gnuradio #2637: _runtime_swig.so installed in the wrong place?

argilo, 2019-07-18:
I just installed version 3.8.0.0-rc1 on an Ubuntu 18.04 machine using PyBOMBS, and got the following error when I tried to run gnuradio-companion:

> Cannot import gnuradio.
> [...]
> (No module named '_runtime_swig')

I noticed that `_runtime_swig.so` was installed in `prefix/lib/python3/dist-packages/gnuradio/gr/`, while the corresponding Python files (`runtime_swig.py`, `runtime_swig.pyc`, `runtime_swig.pyo`) were installed in `prefix/lib/python3.6/dist-packages/gnuradio/gr/`.

Once I moved `_runtime_swig.so` from `lib/python3` over to `lib/python3.6`, gnuradio-companion started up normally.

https://github.com/gnuradio/gnuradio/pull/2517 changed the way runtime_swig is installed, so it could possibly be related.

marcusmueller, 2019-07-18:
@argilo  thanks for reporting! We'll investigate.

marcusmueller, 2019-07-18:
We've got reports confirming this, while I'm personally still trying to reproduce.

marcusmueller, 2019-07-18:
with #2640 I think we've merged a patch. I checked, it works on Ubuntu 18.04 without Pybombs. If you're OK with it, I'd declare this issue closed. Are you OK with it?

argilo, 2019-07-18:
@marcusmueller All good now! I did a fresh install with PyBOMBS and `_runtime_swig.so` ended up in the right place. :ok_hand:

closed 2019-07-18 by argilo