gnuradio/gnuradio #2642: fec: cc enc/dec in ber_curve_gen example yields different results in 3.8

mormj, 2019-07-19:
When running the ber_curve_gen.grc example in gr-fec, the convolutional coder results do not look right at all as compared to the graph produced in 3.7

Here is the graph produced under 3.8:
[image]

Under 3.7
[image]

The parameters for the objects in the .grc file all look the same between both

jdemel, 2019-07-20:
It looks like the CC code fails. Besides, the curves seem to be correct. If the simulation would run for a couple more hours, I'd guess, those curves would look the same. I suggest to investigate, why the CC code fails. What about CC code tests? Do they pass?

fmagno, 2020-02-06:
There seems to be an issue with the example `fec/examples/fecapi_cc_decoders.grc` that could be related:
[image]

marcusmueller, 2024-01-03:
Currently broken, can't instantiate Bercurve Sink:

```
Param - Bus_Structure_Sink(bus_structure_sink):
	Value "list(map(lambda b:list(map(lambda a:b * len(esno) * 2 + a, range(len(esno)*2))), range(num_curves)))" cannot be evaluated:
	name 'num_curves' is not defined
```

although `num_curves` is a parameter

marcusmueller, 2024-01-03:
Depends on #7044

argilo, 2024-01-03:
I expect this was due to the broken AVX2 convolutional decoder in VOLK, which was removed in https://github.com/gnuradio/volk/pull/458. I recently tracked down the bugs in https://github.com/gnuradio/volk/pull/736#issuecomment-1868383520.

argilo, 2024-01-06:
To get the BER curve example working again, I applied the workaround described in https://github.com/gnuradio/gnuradio/issues/5627#issuecomment-1060608139, and reverted #6889 to solve #7059.

[image]

The convolutional decoder seems to work fine now, but LDPC appears broken. :cry:

argilo, 2024-01-06:
If I revert #6748, then LDPC appears to work again:

[image]

argilo, 2024-01-07:
I'll close this off, since the original bug with the convolutional decoder was fixed in https://github.com/gnuradio/volk/pull/458.

I opened #7060 to cover the new issue with the LDPC decoder.

closed 2024-01-07 by argilo