gnuradio/gnuradio #2653: grc 3.8: wrong updating of I/O color hint after block I/O type change

Notou, 2019-07-22:
Hello everyone.

I was testing the new 3.8 and I noticed that, when using the arrow keys to change the I/O type of a block with more than one input or output, only the first one gets updated to the color of the new type. The other ones stay the same.
This can lead to pretty strange graphs:

[image]

Opening and closing the block's parameters (even with clicking cancel) updates the I/O types correctly, but it's a bit tedious and can be misleading.

argilo, 2019-07-27:
I encountered the same bug this morning. It seems worth fixing before the final 3.8 release.

dkozel, 2019-07-27:
@Notou Thanks for the report. If this is a change from 3.7 then I agree it's a bug worth fixing before the release. If it isn't a change then it should be a fix in the 3.8.0.1 release which won't be far behind.

argilo, 2019-07-27:
In 3.7 all ports change colour when the up/down keys are used.

argilo, 2019-08-05:
I spent a bit of time digging into this today but didn't find the root cause. I suspect it's an issue with the `dtype` attribute not being tracked / updated properly for [`PortClone`](https://github.com/gnuradio/gnuradio/blob/master/grc/core/ports/clone.py) instances.

/cc @skoslowski

glennlive, 2019-09-20:
This should be fixed by this commit...

https://github.com/gnuradio/gnuradio/commit/9a193b3cd25d4a2411511d9ceed667944d8b4e3a

argilo, 2019-09-24:
Thanks @glennlive! That does appear to fix the bug. The change still needs to be included in the maint-3.8 branch though.

maitbot, 2020-11-15:
Fixed in 813412a1d020947b5062cfae8304a998b21d1be6
This Issue can be closed.

closed 2020-11-16 by Notou