gnuradio/gnuradio #2658: FE Corrections not working in UHD: USRP Source

erikheinz, 2019-07-23:
Hi,
it seems to me, the controls "Enable DC Offset Correction" and "Enable IQ Imbalance Correction" in the grc block **UHD: USRP Source** are not functional any more in gnuradio-3.8. The controls are there, but they have no effect.

I guess there should be something like
`self.${id}.set_auto_iq_balance(${iq_imbal_enb0}, 0)`
`self.${id}.set_auto_dc_offset(${dc_offs_enb0}, 0)`
 in the make and the callbacks sections of uhd_usrp_source.block.yml, shouldn't it?

By the way, why are these parameters defined as "dtype: raw", not "dtype: bool"?

Best regards,
Erik

haakov, 2019-07-23:
Hello Erik! :)

There is a pull request for this issue: https://github.com/gnuradio/gnuradio/pull/2595

marcusmueller, 2024-01-03:
Fixed in #3021

closed 2024-01-03 by marcusmueller