gnuradio/gnuradio #2665: `check_topology` and `forecast` should be non-state-modifying

marcusmueller, 2019-07-24:
i.e. they need to get the `const` (as in `void block::forecast() const`) qualifier, so that people can't use them to abuse them and assume that the last call to `check_topology` defines the number of in- and output streams that will be used. This should be queryable at `start()` and `work()` time, maybe?

Same goes for `forecast`.

Motivation: Make it clearer where block state is supposed to be modifiable and where not. This is essentially crucial if we want to allow for scheduling to detach the block executing thread / worker from the managing thread/worker.

marcusmueller, 2019-07-25:
There might simply be more methods of `block` that should be considered here:

- [ ] `check_topology`
- [ ] `forecast`
- [ ] `nitems_read` / `nitems_written`
- [ ] `get_tags_in_range`

michaelld, 2019-07-25:
Love the idea (again) ... wondering how much breakage we'll have to fix by tweaking methods that currently not `const` but should be...? Guessing there will be some breakage ... especially If we change `check_topology`: We'll break some GR IT (in-tree) and OOT blocks that use it to set the actual topology ... I think `gr-audio` does this, as do some others. We need to find a better way to setup initial topology!

marcusmueller, 2024-01-03:
- `check_topology` seems to simply need to get documented that it's being used for actual modifications, not checking
- `forecast` should be `const` but there's a solid chance that breaks the ecosystem
- `nitems_read,written` should be `const` (and is not a significant API difference)
- `get_tags_in_range` needs more thought