gnuradio/gnuradio #2668: grc: yaml parser errors prevent startup of grc

mormj, 2019-07-24:
In 3.8, yaml parser errors from grc blocks prevent grc from opening up altogether, rather than just failing the loading of those blocks

Steps to reproduce:
1) create a new block in an OOT
```
josh@josh-p51:~$ gr_modtool newmod breakstuff
Creating out-of-tree module in ./gr-breakstuff...
Done.
Use 'gr_modtool add' to add a new block to this currently empty module.
josh@josh-p51:~$ cd gr-breakstuff/
josh@josh-p51:~/gr-breakstuff$ gr_modtool add -t sync -l python foo
GNU Radio module name identified: breakstuff
Language: Python
Block/code identifier: foo
Please specify the copyright holder: 
Enter valid argument list, including default arguments: 
 
Add Python QA code? [Y/n] n
Adding file 'python/foo.py'...
Adding file 'grc/breakstuff_foo.block.yml'...
Editing grc/CMakeLists.txt... 
josh@josh-p51:~/gr-breakstuff$ mkdir build
josh@josh-p51:~/gr-breakstuff$ cd build
josh@josh-p51:~/gr-breakstuff/build$ cmake ../
```
(fix compilation errors, then .. )
```
josh@josh-p51:~/gr-breakstuff/build$ make install
```
Now, launching grc fails
```
josh@josh-p51:~/gr-breakstuff/build$ gnuradio-companion 
/usr/lib/python3/dist-packages/apport/report.py:13: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
  import fnmatch, glob, traceback, errno, sys, atexit, locale, imp
Traceback (most recent call last):
  File "/share/gnuradio/grdev/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/share/gnuradio/grdev/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/grc/main.py", line 83, in main
    platform.build_library()
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/grc/core/platform.py", line 154, in build_library
    data = cache.get_or_load(file_path)
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/grc/core/cache.py", line 66, in get_or_load
    data = yaml.safe_load(fp)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 94, in safe_load
    return load(stream, SafeLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 72, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 35, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python3/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python3/dist-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 115, in check_token
    while self.need_more_tokens():
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 149, in need_more_tokens
    self.stale_possible_simple_keys()
  File "/usr/lib/python3/dist-packages/yaml/scanner.py", line 289, in stale_possible_simple_keys
    "could not find expected ':'", self.get_mark())
yaml.scanner.ScannerError: while scanning a simple key
  in "/share/gnuradio/grdev/share/gnuradio/grc/blocks/breakstuff_foo.block.yml", line 9, column 1
could not find expected ':'
  in "/share/gnuradio/grdev/share/gnuradio/grc/blocks/breakstuff_foo.block.yml", line 10, column 31
```

In addition to fixing the underlying issue in GRC with flagging parser errors rather than bombing out dramatically, modtool should be fixed to generate more valid grc files as well, as there are some residual xml comments and syntactically incorrect things from the generation.

For reference, here is the current default generated .grc file:
```
id: breakstuff_foo
label: foo
category: '[breakstuff]'

templates:
  imports: import breakstuff
  make: breakstuff.foo(${})

<!-- Make one 'parameters' list entry for every Parameter you want settable from the GUI.
     Sub-entries of dictionary:
     * id (makes the value accessible as \$keyname, e.g. in the make entry)
     * label
     * dtype -->
parameters:
- id: ...
  label: ...
  dtype: ...
- id: ...
  label: ...
  dtype: ...

<!-- Make one 'inputs' list entry per input. Sub-entries of dictionary:
     * label (an identifier for the GUI)
     * domain
     * dtype
     * vlen
     * optional (set to 1 for optional inputs) -->
inputs:
- label: ...
  domain: ...
  dtype: ...
  vlen: ...

<!-- Make one 'outputs' list entry per output. Sub-entries of dictionary:
     * label (an identifier for the GUI)
     * dtype
     * vlen
     * optional (set to 1 for optional inputs) -->
- label: ...
  domain: ...
  dtype: !-- e.g. int, float, complex, byte, short, xxx_vector, ...--

file_format: 1
```

haakov, 2019-07-29:
I guess moving this line down to the try block @ line 168 can be an option, I don't immediately see why it needs to be called before the if-elif-else statements
https://github.com/gnuradio/gnuradio/blob/c184e1f863d9425598cce1de01e6989b2d10dcb1/grc/core/platform.py#L154

mormj, 2019-07-30:
Yes, that seems to do the trick without any apparent side effects.

PR #2690 (2019-08-05): grc: prevent yaml parser errors from crashing grc
Moves the yaml file load inside the try block as recommended by @haakov 

Allows grc to open up when there is a bad yml file in an OOT

Fixes #2668
commit 86b86a9d (2019-08-05): grc: prevent yaml parser errors from crashing grc

Moves the yaml file load inside the try block

Fixes #2668
closed 2019-08-05 by marcusmueller