gnuradio/gnuradio #2672: gr_modtool rm deletes too much

ast, 2019-07-27:
I had two modules `stream_to_vector_decimator` and `stream_to_vector_decimator2`

`gr_modtool rm stream_to_vector_decimator`

removed both of them.

solomonbstoner, 2021-06-01:
I am able to reproduce this behavior with similarly named blocks. I will try to debug this

solomonbstoner, 2021-06-01:
Just to illustrate one of the cause(s) of the problem, let's look at the `rm` command. The `stream_to_vector_decimator2` and `stream_to_vector_decimator` are both deleted because the function `_run_subdir` finds all the files in the path of interest, and offers to delete files where the user entered name appears. So entering  "stream_to_vector_decimator" would match both "stream_to_vector_decimator" and "stream_to_vector_decimator2". That's why files from both blocks were offered to be deleted. 

```python
    def _run_subdir(self, path, globs, makefile_vars, cmakeedit_func=None):
        """ Delete all files that match a certain pattern in path.
        path - The directory in which this will take place
        globs - A tuple of standard UNIX globs of files to delete (e.g. *.yml)
        makefile_vars - A tuple with a list of CMakeLists.txt-variables which
                        may contain references to the globbed files
        cmakeedit_func - If the CMakeLists.txt needs special editing, use this
        """
        if self.cli:
            from ..cli import cli_input
        # 1. Create a filtered list
        files = []
        for g in globs:
            files = files + sorted(glob.glob(f"{path}/{g}"))
        files_filt = []
        logger.info(f"Searching for matching files in {path}/:")
        for f in files:
```
Now, `gr_modtool` will ask you to confirm if you want to delete the files, so a little careful attention could avoid unnecessary deletion. But I suppose we could still improve on the matching nonetheless

solomonbstoner, 2021-06-01:
@mormj assign this issue to me ill try to work something out

PR #4709 (2021-06-25): modtool: Fix #2672 for `rm`, `disable`, `rename` and `makeyml`
Fixes #2672 

When a block name is entered as an argument to a modtool command, (e.g. `gr_modtool rm my_block`), the user expects _only_ that specific block (ie `my_block`) to be acted upon. Issue #2672 is caused when similarly named block names are also acted upon (e.g. `my_block_1` or `temp_my_block`). This issue affects the four commands stated in the title. PR fixes said issue.

Edit:
PR also fixes two other seemingly unreported issues:
1. A minor AttributeError affecting the `disable` command OOT version 38 due to what I belie [... cut]
closed 2021-06-25 by mormj