gnuradio/gnuradio #2684: Update the OOT documentation/tutorial adding the new yml tags to describe grc blocks

OtavioLara, 2019-07-29:
I am not finding in the gnuradio tutorials or even on the internet any kind of explanation about the new tags found in yml files in gnuradio 3.8 or 3.8+.

Example: What is the "domain" at the input of a block:

`
inputs:
- label: in
  domain: ?
  dtype: float
  vlen: 1
`

mhostetter, 2019-07-31:
From reviewing the source code of other's OOT modules, I've determined the `domain` is `stream` for streaming ports and `message` for message passing ports. There may be another value for tagged streams, but I'm not sure.

Notou, 2019-08-09:
I came across [this page](https://wiki.gnuradio.org/index.php/YAML_GRC) on the wiki.
It's not yet linked to anything so it's a bit hard to find.

closed 2024-01-03 by marcusmueller