gnuradio/gnuradio #2687: GRC color scheme help

na0b, 2019-07-30:
I am running GRC on a Live Linux Parrot OS distribution. Many of the text colors are messed up. When editing a block, sometimes the text and the background are of the same color, so you cannot see what you are running. While I am able to change colors in other applications, nothing seems to work for GRC.
Can someone point me in the right direction?
Thanks,
Juan.

mormj, 2019-07-30:
Would you be able to post a screenshot to give a better idea what you are seeing?

na0b, 2019-07-31:
OK.Here is a screen capture. In the ID field of the Variable block, both the text and the background are set to the same color. Thanks.

[image]

mormj, 2019-08-02:
Hmmm .. does changing your color scheme in your linux distro have any effect?

na0b, 2019-08-02:
This is not the case  with the Parrot distro. I  also posted a question at the Parrot community, but received no answer. I will go back to Ubuntu. Thanks.
What I think would be very useful to schools is a lean live GNURadio distro with networking. If you  know of any, please let me know.

cbdonohue, 2019-10-31:
You can change the theme in Parrot OS by going to System -> Look and Feel -> Appearance. Then try a theme like "TraditionalOK" and you should be able to see everything in gnuradio. 
[image]

nlitsme, 2023-10-02:
I noticed that I needed to change my OS color scheme from dark to standard, and then back to dark.
Now the foreground text color in GRC changed from white ( unreadable ) to black ( ok ).
While also retaining the general scheme I like for the rest of my apps.

But since GRC is the first app I found which seems to make a wrong choice by default for the text color, I think the problem may be with GRC.

I am running a kde/plasma desktop on debian.

willem

cafeclimber, 2023-10-02:
I honestly think that's clearer in the long term at least for the GTK version. I personally never even made the connection between the colors and the parameter's type 😅 For Qt, I know you can have different validators on the various fields

marcusmueller, 2023-10-02:
connections between colors and types: me neither, the pastel goth version of the in/output port colors just doesn't strike me as too significant, and also, there's no "string" input port, and reserving anything reddish for "invalid" seems just right…

marcusmueller, 2023-10-03:
That's not bad, and should you have already coded this: I'd advocate we try it, see how people using `main` like it, because it's already much better (imho) than the staus quo, though I'd have gone for more of a tabular approach, with an illustrative third column in that layout

|  |   |   |
|---|---|---|
| Max Number of Output | `[ Text entry field ]` | ℤ integer |
| Author | `[ Text entry field ]` | A…Z string |
| Height of Respect | `[ Text entry field ]` | ℝ float |
| Wave number | `[ Text entry field ]` | ℂ complex |
| Favorite Monarch | `[ Count von Count ▾ ]` | choice |
| Rage against the Machine | `[ On ▾ ]`  | ✓✗boolean |
| QSS Theme | `[ text field ] [File selector btn]` | 📂 File for reading |

(also, while I'm looking at dialogs… can we not center-align the labels?)

cafeclimber, 2023-10-03:
Yeah this was a very simple edit of a couple lines. Splitting into three columns shouldn't be too difficult? And I'm happy to switch to left-aligned lol

chrboesch, 2023-10-12:
@cafeclimber Thanks, that helps me a lot! :smile:

willcode, 2023-10-12:
Addressed by #6888 #6911

closed 2023-10-12 by willcode