gnuradio/gnuradio #2688: grc: fails to build simple c++ based flowgraph

dl1ksv, 2019-07-30:
Trying to build a simple flowgraph in c++ [test_cpp](https://github.com/gnuradio/gnuradio/blob/master/grc/tests/resources/test_cpp.grc)

fails with:

 Executing: cmake .. && make &&  /usr/bin/xterm -e /home/schroer/gnuradiocomponents/gnuradio/grc/tests/resources/horn/build/horn

-- Configuring done
-- Generating done
-- Build files have been written to: /home/schroer/gnuradiocomponents/gnuradio/grc/tests/resources/horn/build
Scanning dependencies of target horn_autogen
[ 25%] Automatic MOC for target horn
[ 25%] Built target horn_autogen
Scanning dependencies of target horn
[ 50%] Building CXX object CMakeFiles/horn.dir/horn.cpp.o
In file included from /home/schroer/gnuradiocomponents/gnuradio/grc/tests/resources/horn/horn.cpp:8:
/home/schroer/gnuradiocomponents/gnuradio/grc/tests/resources/horn/horn.hpp:13:10: fatal error: gnuradio/top_block.h: Datei oder Verzeichnis nicht gefunden
 #include <gnuradio/top_block.h>
          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/horn.dir/build.make:63: CMakeFiles/horn.dir/horn.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/horn.dir/all] Fehler 2
make: *** [Makefile:84: all] Fehler 2

Done (return code 2)

haakov, 2019-07-30:
Yeah the CMakeLists Mako template needs to be updated to modern CMake. I've begun working on it, but I'm not quite finished yet.

haakov, 2019-07-30:
Note to self: Give that flowgraph another ID :)

dkozel, 2020-04-05:
@haakov I just tested this on the head of `maint-3.8` and hit the same error. Would you be able to work on this?

dkozel, 2020-04-05:
Tested on `master`, same error.

haakov, 2020-04-11:
Yep, I'll look into it.

kc2rxo, 2021-05-31:
Just adding that I am running into this as well. Installed on a clean Ubuntu 20.04 using apt. Version from GRC is 3.8.1.0 (Python 3.8.5).

I do not get the error above when running the test_cpp. Instead I get `sequence item 2: expected str instance, NoneType found`.

Any updates on a fix?

dl1ksv, 2021-06-01:
I just tried to run he example test_cpp.grc in the master branch.

It fails with:

/home/schroer/gnuradiocomponents/errors/horn/horn.hpp:17:10: schwerwiegender Fehler: QVBoxLayout: Datei oder Verzeichnis nicht gefunden
   17 | #include \<QVBoxLayout>
      |          ^~~~~~~~~~~~~
as 

gnuradio::gnuradio-qtgui

is missing in CMakeLists.txt in the target_link_libraries section.

After adding this compiling fails with 
/home/schroer/gnuradiocomponents/errors/horn/horn.cpp:62:22: Fehler: Variable »QApplication app« hat Initialisierung, aber unvollständigen Typ
   62 |     QApplication app(argc, argv);
      |                      ^~~~

as 
#include \<QApplication>
is missing in horn.hpp.

After adding this statement I can build and run the flowgraph.

I'll have a look , if I can make a pr to fix this, but I suppose this won't fix your issue. Can you post your flowgraph ?

kc2rxo, 2021-06-03:
@dkozel to answer your question it is this one [here](https://github.com/gnuradio/gnuradio/blob/master/grc/tests/resources/test_cpp.grc).

So after a bit of fiddling about I got further along. I uninstalled gnuradio via apt, added the PPA for it and reinstalled. This put me on version 3.9.1.0 (Python 3.8.5). I get a warning about xterm which appears and disappears on first build. Now the file is failing to build similar to above, just different packages are missing I think.

I specifically got this:

```
Executing: cmake .. && cmake --build . -j1 && cd ../.. && None -e /home/adam/horn/build/horn

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found LOG4CPP: /usr/lib/x86_64-linux-gnu/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   Found gmp, version 6.2.0
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Found Boost: /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.71.0") found components: date_time program_options filesystem system regex thread unit_test_framework 
-- Found Volk: Volk::volk  
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.8.5") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.8.so (found suitable exact version "3.8.5") 
-- Checking for module 'sndfile'
--   No package 'sndfile' found
-- Could NOT find SNDFILE (missing: SNDFILE_LIBRARIES SNDFILE_INCLUDE_DIRS) 
-- Checking for module 'fftw3f >= 3.0'
--   Found fftw3f , version 3.3.8
-- Found FFTW3f: /usr/lib/x86_64-linux-gnu/libfftw3f.so  
-- Checking for module 'alsa'
--   Found alsa, version 1.2.2
-- Found ALSA 1.2.2
-- Checking for module 'jack'
--   No package 'jack' found
-- Checking for module 'portaudio-2.0'
--   No package 'portaudio-2.0' found
-- Checking for module 'Qt5Qwt6'
--   Package 'Qt5Svg', required by 'Qt5Qwt6', not found
-- QWT Version: 6.1.4
-- Found Qwt: /usr/lib/libqwt-qt5.so  
-- Checking for module 'uhd'
--   No package 'uhd' found
-- Could NOT find Codec2 (missing: LIBCODEC2_LIBRARIES LIBCODEC2_INCLUDE_DIRS) 
-- Found GSM: /usr/lib/x86_64-linux-gnu/libgsm.so  
-- Checking for module 'libzmq'
--   No package 'libzmq' found
-- Could NOT find ZeroMQ (missing: ZEROMQ_LIBRARIES ZEROMQ_INCLUDE_DIRS) 
-- Configuring done
CMake Error at CMakeLists.txt:38 (add_executable):
  Target "horn" links to target "sndfile::sndfile" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
```

After installing the missing package: `sudo apt install libsndfile1-dev` I wound up where @dkozel was on June 1st. Following his steps I got the flowgraph to build by executing the commands outside of GRC (running the build/run in GRC causes the CMakeList file to be rewritten?). I ran the flowgraph off the command line to be greeted with a blank window - which makes sense.

I am going to try another graph now with more than just null sink/sources.

dl1ksv, 2021-06-04:
After applying #4710 you can generate and run the above flowgraph from grc.

But after  changing the null source to a signal source the build again fails.
There is still much to do to generate useful cpp code from grc.

dl1ksv, 2021-06-04:
Just added another commit in #4710 so signal source and audio sinks and other blocks are correctly build and run in cpp

dl1ksv, 2021-06-21:
Fixed with #4710

closed 2021-06-21 by dl1ksv