gnuradio/gnuradio #2692: gnuradio-companion with USRP N210 - Port is not connected

Titi9237, 2019-07-31:
Hi everyone,

I am just beggining with gnuradio and an USRP N210 that I would like to use. After installing gnuradio and uhd on my Ubuntu 18.04, I can connect the N210 using an Ethernet connection (1GB/s), the uhd_find_device and uhd_usrp_probe commands are successfull.

But when I open gnuradio and I try to use a block "UHD: USRP Source" (following this [tutorial](https://www.youtube.com/watch?v=KWeY2yqwVA0)) I end up with a connection error.
> Error 0:
> Block - uhd_usrp_source - UHD: USRP Source(uhd_usrp_source):
>   Source - out(0):
>     Port is not connected.

[image]

What am I doing wrong ? 
Thanks

mormj, 2019-07-31:
The better place for this type of discussion is the mailing list:
https://wiki.gnuradio.org/index.php/MailingLists
or the GNU Radio slack/IRC

But I would have to see the flowgraph in grc to see why the port is not connected

Titi9237, 2019-07-31:
It was an awful noob mistake.
Sorry for the inconvenience...

closed 2019-07-31 by Titi9237

mashfique1, 2025-10-03:
@Titi9237 what fixed it? what was the noob mistake? some how i had the same error and but went away after restarting. I am not sure if same happened with you