gnuradio/gnuradio #2696: pfb_clock_sync_ccf_impl.cc crashes

armleo, 2019-08-02:
So I was creating a modem using pfb clock sync, but when connected my blocks together my RX side begun to crash. I debugged my project by placing Nulls sinks and Null sources instead of blocks to see what happens and crash persisted until I replaced PFB Clock Sync.

I removed everything and connected a noise source to PFB Clock Sync's input and connected output to null sink. Now it still crashes but with random intervals in 1 minute to 20 hours instead of a couple of seconds in original project.

Here are the blocks:
Noise Source -> PFB Clock Sync -> Null Sink
PFB Clock sync uses these parameters
sps: 2
loop bandwidth 6.28/100
taps: filter.firdes.root_raised_cosine(
            32,          # gain
            32,          # sampling rate based on 32 filters in resampler
            1.0,         # symbol rate
            0.5,         # excess bandwidth (roll-off factor)
            11*sps*32)
filter size: 32
initial phase: 16
Maximum rate deviation: 1.5
output sps: 1

How i can debug these issue. I suspect these parts in original source code to access "in" pointer in inecorrect place:
432: int count = 0;
459: while(d_filtnum < 0) {
	    d_k += d_nfilters;
	    d_filtnum += d_nfilters;
	    count -= 1;
	  }
503: gr_complex diff = d_diff_filters[d_filtnum]->filter(&in[count]);

nickoe, 2020-04-01:
@armleo Hi, thank you for the report, but could you attach the grc file and state which version of gnuradio you are running?

armleo, 2020-07-24:
The issue was that the block accepted NAN and Infinity, it caused strange bugs.

The issue with noise source connected to input was different issue, causing some internal variables to behave strange.

Closing this, as the issue was not this block.

closed 2020-07-24 by armleo