gnuradio/gnuradio #2700: Compilation Error!!

r00tb3, 2019-08-03:
I'm building GNUradio and facing the same compilation errors on different branches.

```
$ git branch 
* maint-3.7
  maint-3.8
  master

$ make -j4 V=1
[  0%] Built target pygen_volk_python_volk_modtool_5b5d0
.
.
.
[ 32%] Built target gnuradio-fft
[ 32%] Built target gr_fec_rstest
/usr/bin/ld: /usr/local/lib/libpython3.6m.a(abstract.o): relocation R_X86_64_32S against _Py_NotImplementedStruct can not be used when making a shared object; recompile with -fPIC
/usr/local/lib/libpython3.6m.a: error adding symbols: Bad value
collect2: error: ld returned 1 exit status
gnuradio-runtime/swig/CMakeFiles/pmt_swig.dir/build.make:99: recipe for target 'gnuradio-runtime/swig/_pmt_swig.so' failed
make[2]: *** [gnuradio-runtime/swig/_pmt_swig.so] Error 1
CMakeFiles/Makefile2:1305: recipe for target 'gnuradio-runtime/swig/CMakeFiles/pmt_swig.dir/all' failed
make[1]: *** [gnuradio-runtime/swig/CMakeFiles/pmt_swig.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 37%] Built target gnuradio-uhd
[ 44%] Built target gnuradio-audio
[ 47%] Built target gnuradio-blocks
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2 
```
I'm not able to debug the error.

What's the error about exactly?

How do I solve it?

Also, I want to build for the version 3.8.0 but I'm facing the same error.

Pls, help!!

Regards.

michaelld, 2019-08-03:
Can you tell us the OS & version, how you installed background dependencies, and what version of SWIG is installed?

r00tb3, 2019-08-03:
> Can you tell us the OS & version, how you installed background dependencies, and what version of SWIG is installed?

```
OS & Version:-
Linux XXX 4.15.0-47-generic #50~16.04.1-Ubuntu SMP x86_64 x86_64 x86_64 GNU/Linux
```

Background Dependencies:-
Followed this post:- [Gnuradio-Wiki](https://wiki.gnuradio.org/index.php/UbuntuInstall#Xenial_Xerus_.2816.04.29)

```
$ swig -version

SWIG Version 3.0.8

Compiled with g++ [x86_64-pc-linux-gnu]

Configured options: +pcre
```

michaelld, 2019-08-03:
Just noticed 2 things: (1) the Python is installed into `/use/local` which I’m guessing means installed from source & not via `apt-get`. CMake isn’t very robust at finding the correct related Python libraries & headers ... you’re best off specifying these directly via the appropriate CMake arguments. (2) GNU Radio 3.7 requires Python 2.7; it can’t build using any newer version.

r00tb3, 2019-08-04:
@michaelld I have Python 2.7, 3, 3.5, 3.6.9 installed.

How do I pass an argument specifying which python version to use?

Also, how can I install GNUradio 3.8.0 in facing the same problems for it?

Regards.

r00tb3, 2019-08-05:
I installed gnuradio using Pybombs.

Installing from the source never worked for me.

closed 2019-08-05 by r00tb3

michaelld, 2019-08-05:
@r00tb3 Glad you got it working! pybombs does work pretty well for GR3.7 ... not sure about GR3.8 support being robust yet. Good luck with your GR work!

mgarrett1955, 2019-10-01:
Hello, I am having issues using pybombs to install 3.8. I am installing into a fresh Ubuntu 18.4, I get the following error: 

-- Python checking for numpy - not found
-- 
-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Dependency PYTHON_MIN_VER_FOUND = TRUE
--   Dependency PYYAML_FOUND = TRUE
--   Dependency MAKO_FOUND = TRUE
--   Dependency PYGI_FOUND = TRUE
--   Dependency GTK_GI_FOUND = TRUE
--   Dependency CAIRO_GI_FOUND = TRUE
--   Dependency PANGOCAIRO_GI_FOUND = TRUE
--   Dependency NUMPY_FOUND = FALSE
CMake Error at cmake/Modules/GrComponent.cmake:75 (message):
  user force-enabled gnuradio-companion but configuration checked failed
Call Stack (most recent call first):
  grc/CMakeLists.txt:95 (GR_REGISTER_COMPONENT)

As described here, it seesm to not be finding numpy, and I am using pybombs to install 3.8. I believe that it is installing Python 2.7.