gnuradio/gnuradio #2718: Integrate soapysdr

marcusmueller, 2019-08-10:
GNU Radio should be like Python – not in the being a dynamically typed language kind of way, but in the coming with batteries included kind of way.

devnulling, 2020-01-23:
@guruofquality Trying to round up some folks who would like to work on PRs. Is this of any interest to you (merging https://github.com/pothosware/gr-pothos ) ?

guruofquality, 2020-02-03:
> @guruofquality Trying to round up some folks who would like to work on PRs. Is this of any interest to you (merging https://github.com/pothosware/gr-pothos ) ?

I'm waiting on some 3.8 changes for GRC, but yea. That could save on some copy pasta and version management. One benefit is that this project may be only one that uses block_executor.h outside of the tree. Since its not installed, so its not easy for out of tree projects ;-) On the other hand there are often changes and additions to this project and I don't want to bother maintainers.

devnulling, 2020-02-03:
@guruofquality Sorry I had accidentally copied in the wrong link (gr-pothos) and had meant to copy in the gr-soapy link [ https://gitlab.com/librespacefoundation/gr-soapy/ ]

ncorgan, 2020-04-19:
Given the duplicate functionality and GNU Radio's trend of moving away from hardware-specific components, would this be planned to replace gr-uhd or live alongside it?

devnulling, 2020-04-22:
@ncorgan The idea is to leave gr-uhd in tree, and gr-soapy will live along side it. There is also work being done to bring gr-iio in tree (https://github.com/gnuradio/gnuradio/pull/2840)

mrjacobagilbert, 2020-04-22:
To offer an unsolicited opinion on this:

I'll pick on gr-uhd because i have been dealing with this for many years. Having something like UHD that has its own dependencies and libraries that are separately evolving located within GR is a hassle compared to having it OOT. When its in-tree, when any deps under gr-uhd (e.g.: libuhd) change GR must be updated. If it were OOT, so long as gr-uhd remains compatible, the hardware interfaces/drivers/etc, which aren't needed by GR, can change as they need to and just UHD can be updated.

I'm not quite as familiar w/ soapy as IIO, but from my perspective, pulling more edge-interface stuff in-tree is making GR less modular, not more.

guruofquality, 2020-04-22:
> I'm not quite as familiar w/ soapy as IIO, but from my perspective, pulling more edge-interface stuff in-tree is making GR less python-like, not more.

Its not necessarily a bad idea. Most gr-projects could be hosted out of tree either with the specific project maintainer, with the hardware manufacturer's repositories, or possibly on gnuradio's github itself -- with appropriate permission management of course. After-all, there is no shortage of github and gitlab repositories and accounts. Something like that would probably make it easier to just maintain gnuradio core libraries itself.

Of course more out of tree projects mean that there is usually more stuff to install to get basic examples going. I think pybombs and other installation methods can mitigate that. Also, package maintainers will have to make more packages for these high priority out of tree projects. And synchronizing API changes across packages is already enough "fun" for the community, more out of tree of course means more of that. :-)

In terms of promoting out of tree, maybe gnuradio should offer official hosting for packages on https://github.com/gnuradio and or maintain a list of official maintainers/hosting for of out of tree packages -- im sure that's already done to some extent but I have personally got thrown off by highly venerable forks.

Personally I would lean towards more out of tree when possible. Anyway, I don't have a horse in this race, I with gr-soapy the best and I will try to keep the SoapySDR API changes to a minimum.

mrjacobagilbert, 2020-04-22:
Agree with what you are saying here. Just looking at the two official pybombs recipe files, many (most?) of the projects don't build under 3.8.

Some sort of official listing of high quality OOTs that are maintained to GR code standards and track GR development would be a really good approach as far as I'm concerned.

marcusmueller, 2020-04-25:
so, both of you are, in my opinion, right:

1. We're not trying to pull _everything under the sun_ into GNU Radio itself. That'd make no sense, unless the maintainer was longing for everlasting pain (he's not). What GNU Radio is, and should be, is an ecosystem of things that you can use to build SDR applications. That means that we should encourage good out of tree work.
2. There's things that are system-critical – you wouldn't want to move gr-fft/gr-filter out of tree ("in-tree" here means: maintained, CI'd and released with GNU Radio releases), even if it uses a special dependency (fftw) that not many other modules use – it's just so fundamental to the ecosystem that we need to both guarantee quality standards and stability, and be 100% in the testing loop with whatever we change upstream.

The same consideration applies to universal hardware interfaces: up to now, a lot of especially non-institutional users install GNU Radio, then install gr-osmosdr to be able to do anything useful with their RTL dongles and hackrfs. That needs to change; and that's what I meant with "batteries included":

Not necessarily that we deliver your torch with the batteries already inserted, (I hope we'll see finely granular packaged GNU Radio in the future more often, working on that architecturally), but that we care about supplying basic batteries alongside with the torch, and having tested the torch with actual batteries.

Frankly, soapysdr is a relatively benign dependency. Plugin arch means packages can package a minimal core component, and the hardware support for your favourite device comes in a separate binary package.

And, Josh is right, it would totally make our life easier: so far, we can't really tell people in the tutorials to just get their darn RTL-dongle, and plug it in and look for FM stations. And that's what a pretty large amount of our users want to do! This leads to a lot of support work behind the scenes: writing emails explaining how to install gr-osmosdr (we're not even taking the brunt of that work – that's on the Osmocom folks, who could be contributing to the ecosystem in that time!), making sure pybombs can build that on any distro, fiddling with pecularities of things out of project control (again, the osmocom people are great, not complaining)...

In other words, if you go take a hike and feed your neighbor's cow every day so you can milk her, at some point it might be easier to have your own cow. Luckily, free software cows make that easy :)

willcode, 2021-02-16:
I've done a port to 3.9 (https://gitlab.com/willcode4/gr-soapy) but am not sure how we want to proceed, and if it's better to coordinate with librespacefoundation.

willcode, 2021-03-06:
In-tree WIP: https://github.com/willcode/gnuradio/tree/gr-soapy-in-tree builds and runs. More work to be done, but will definitely be ready for v3.10.

Reading Marcus' comment above, I should have named this branch own-cow.

5ch4um1, 2021-03-27:
any news on this one? @willcode ?
https://github.com/willcode/gr-soapy gives me errors on my Lubuntu 18.04 Box, tried to follow the instructions, but when running make i get a lot of errors like:
In function `pybind11_static_set':
<artificial>:(.text+0x82): undefined reference to `PyProperty_Type'

So if i want to actually use my "rtlsdr or hackrf", the best option seems to be a downgrade to 3.8 and wait for 3.10?
Does that in-tree fork also take hours to build? I'm still a bit traumatized by compling 3.9 for about 6 hours just to realize that neither osmosdr nor soapysdr seem to be easily available, so my shiny new gnuradio 3.9 is pretty much useless? Own cow or batteries included would be really nice to make gnuradio more accessible for average people?

willcode, 2021-03-28:
Yes, I'm actively working on the in-tree version. I expect to have this in the master branch soon, and in the June release of 3.9. Here is the work-in-progress rebased onto both master and maint-3.9. If you are up for another compile, the source does work. I run the airspyhf and sdrplay drivers occasionally while developing.

https://github.com/willcode/gnuradio/tree/feature/master/gr-soapy
https://github.com/willcode/gnuradio/tree/feature/maint-3.9/gr-soapy

Make sure you have SoapySDR installed before configuring gnuradio, and make sure it says gr-soapy is enabled. Otherwise, you'll have another long compile for not much gain.

What are you compiling on that takes 6 hours?

Feel free to follow along and test it out if you can.

5ch4um1, 2021-03-28:
@willcode tried https://github.com/willcode/gnuradio/tree/feature/maint-3.9/gr-soapy and so far it seems to work,
thank you very much! :)
[image]

willcode, 2021-03-28:
Thanks for testing, and glad it is working for you. The API and GRC blocks are not nailed down yet, so things may require some tweaking when this is final. There will be SDR-specific blocks for one thing. The single block is pretty much unmaintainable.

5ch4um1, 2021-03-30:
By the way, it seems in January there was a commit in gr-osmosdr and this works now as well for 3.9, just had to install the latest librtlsdr, so now i have 2 source blocks to choose from, thanks again. :)
Would be really nice to have at least a better timing with the releases, like, have at least one source block apart from the uhd ready so 'normal people' can actually use it. In tree would obviously be practical. OOT has advantages as well i guess, but having at least something that works would probably be less disappointing for new users?

willcode, 2021-03-30:
Yes, that's the purpose of bringing gr-soapy in tree. It won't be practical to bring gr-osmosdr in tree because it contains hardware-specific code.

mbr0wn, 2021-05-14:
Closed by https://github.com/gnuradio/gnuradio/pull/4536.

closed 2021-05-14 by mbr0wn