gnuradio/gnuradio #2719: Identify desirable dependency version increases

marcusmueller, 2019-08-10:
GR 3.9 can rely on whatever is state of the art in 2020.

Main Candidates, from the top of my head

* GCC/Clang/MSVC
  * is a C++ version increment to C++14 possible? How far are we from C++17?
* Boost
* CMake
* SWIG

jdemel, 2020-01-23:
I assume those dependency versions depend on distros we'd like to support. Same goes for MacOS and Windows (probably Win10 only?) I'd require those distros to still receive support by whomever works on them. e.g. Ubuntu 14.04 is not a target.
So we should name those distros. I assume Fedora/RedHat/CentOS but which versions?
Ubuntu should be on the list. I suggest 18.04 and 20.04. Other versions are non-LTS or too old. 16.04 is debatable at this point. 

C++17 might be usable with GCC9 in Ubuntu 20.04 but has incomplete support on Ubuntu 18.04 with GCC 7.

jdemel, 2020-05-01:
We move from SWIG to PyBIND11. Thus, we'll drop the SWIG dependency entirely and depend on PyBIND11 instead. I suggest at least PyBIND11 2.2 because there they changed how to add a module.

marcusmueller, 2020-11-07:
re: thrift: #3920

PR #5278 (2021-11-24): Bump Major Dependency Versions for 3.10
# Pull Request Details
---
## Description
For the 3.10 release we will support the latest LTS releases from major distributions.  The major bottleneck here in terms of dependencies is Ubuntu 3.10.

## Related Issue
Fixes #2719

## Which blocks/areas does this affect?
top level CMake

## Testing Done
None

## Checklist

- [x] I have read the [CONTRIBUTING document](https://github.com/gnuradio/gnuradio/blob/master/CONTRIBUTING.md).
- [x] I have squashed my commits to have one significant change per commit. 
- [x] I [have signed my commits before making thi [... cut]
closed 2021-11-24 by mormj