gnuradio/gnuradio #2726: GRC38 OSX +Quartz Runtime Crashing

michaelld, 2019-08-12:
Original thread on gnuradio-discuss is here: https://lists.gnu.org/archive/html/discuss-gnuradio/2019-08/msg00031.html . The basic issue occurs with GRC in GR 3.8.0.0 on OSX when using Quartz for GUI display, not X11. X11 GUI seems to work correctly. The issue is that dragging a block from the righthand list onto a FG will result in a crash.

michaelld, 2019-08-12:
The current potential fix is https://github.com/eblot/gnuradio/commit/62f3065f546813cab754b10b270c8a7af7664a63 .

michaelld, 2019-08-12:
>> We're wondering if this has something to do with retina versus non-retina/4K display ... need a way to enable and disable HiDPI within GTK.

michaelld, 2019-08-12:
MLD> I have plenty of Mac computers running OSX natively for testing: 10.[45] PPC, 10.[5-15] Intel. Some are retina and some aren't. I've never run GR +Quartz, but I'm game to give it a go!

eblot, 2019-08-12:
The crash occurs when an attempt to connect an existing block is made by clicking/releasing a block inlet/outlet.

> The current potential fix is [eblot@62f3065](https://github.com/eblot/gnuradio/commit/62f3065f546813cab754b10b270c8a7af7664a63) .

This one addresses the drag n' drop issue. Drag n' drop issue prevents creating blocks in GRC 3.8 by dragging items from the treeview to the drawing area.  I think this one should be addressed with another ticket.

eblot, 2019-08-13:
GRC backtrace:

    (lldb) bt
    * thread #2, queue = 'com.apple.main-thread', stop reason = signal SIGABRT
      * frame #0: 0x00007fff685cd2c6 libsystem_kernel.dylib`__pthread_kill + 10
        frame #1: 0x00007fff68688bf1 libsystem_pthread.dylib`pthread_kill + 284
        frame #2: 0x00007fff685376a6 libsystem_c.dylib`abort + 127
        frame #3: 0x00007fff6850020d libsystem_c.dylib`__assert_rtn + 324
        frame #4: 0x00007fff3c967aad CoreGraphics`CGGStackRestore + 143
        frame #5: 0x00007fff3c9679fe CoreGraphics`CGContextRestoreGState + 32
        frame #6: 0x00000001123a51ad libcairo.2.dylib`_cairo_quartz_surface_finish + 39
        frame #7: 0x00000001123873e3 libcairo.2.dylib`_cairo_surface_finish + 26
        frame #8: 0x00000001123864af libcairo.2.dylib`cairo_surface_destroy + 97
        frame #9: 0x000000011234ac25 libcairo.2.dylib`_cairo_gstate_fini + 165
        frame #10: 0x000000011234772d libcairo.2.dylib`_cairo_default_context_fini + 60
        frame #11: 0x00000001123478a7 libcairo.2.dylib`_cairo_default_context_destroy + 14
        frame #12: 0x00000001125a2e37 _cairo.cpython-37m-darwin.so`pycairo_dealloc + 23
        frame #13: 0x000000010010ad18 Python`dict_dealloc + 126
        frame #14: 0x0000000100121dd5 Python`subtype_dealloc + 1052
        frame #15: 0x000000010010852d Python`insertdict + 821
        frame #16: 0x000000010010c262 Python`_PyObjectDict_SetItem + 108
        frame #17: 0x0000000100115ff7 Python`_PyObject_GenericSetAttrWithDict + 233
        frame #18: 0x00000001001156fd Python`PyObject_SetAttr + 108
        frame #19: 0x000000010017a04f Python`_PyEval_EvalFrameDefault + 19700
        frame #20: 0x000000010017e7a6 Python`_PyEval_EvalCodeWithName + 1870
        frame #21: 0x00000001000e8a77 Python`_PyFunction_FastCallKeywords + 225
        frame #22: 0x000000010017def7 Python`call_function + 753
        frame #23: 0x0000000100176db3 Python`_PyEval_EvalFrameDefault + 6744
        frame #24: 0x00000001000e8e8c Python`function_code_fastcall + 112
        frame #25: 0x00000001000e9829 Python`_PyObject_Call_Prepend + 150
        frame #26: 0x00000001000e8bbf Python`PyObject_Call + 136
        frame #27: 0x0000000111e3b4a2 _gi.cpython-37m-darwin.so`pygi_signal_closure_marshal + 994
        frame #28: 0x0000000111ff54fd libgobject-2.0.0.dylib`g_closure_invoke + 189
        frame #29: 0x000000011200bad9 libgobject-2.0.0.dylib`signal_emit_unlocked_R + 2521
        frame #30: 0x000000011200c813 libgobject-2.0.0.dylib`g_signal_emit_valist + 2723
        frame #31: 0x000000011200cd32 libgobject-2.0.0.dylib`g_signal_emit + 130
        frame #32: 0x0000000118146568 libgtk-3.0.dylib`gtk_widget_event_internal + 248
        frame #33: 0x0000000117fca3ff libgtk-3.0.dylib`gtk_propagate_event + 255
        frame #34: 0x0000000117fc9dde libgtk-3.0.dylib`gtk_main_do_event + 1294
        frame #35: 0x00000001133744c1 libgdk-3.0.dylib`_gdk_event_emit + 49
        frame #36: 0x00000001133a1bd2 libgdk-3.0.dylib`gdk_event_dispatch + 50
        frame #37: 0x0000000111eb7906 libglib-2.0.0.dylib`g_main_context_dispatch + 310
        frame #38: 0x0000000111eb7c92 libglib-2.0.0.dylib`g_main_context_iterate + 514
        frame #39: 0x0000000111eb7d44 libglib-2.0.0.dylib`g_main_context_iteration + 100
        frame #40: 0x000000011218097d libgio-2.0.0.dylib`g_application_run + 493
        frame #41: 0x0000000112056914 libffi.6.dylib`ffi_call_unix64 + 76
        frame #42: 0x00000001120562cd libffi.6.dylib`ffi_call + 797
        frame #43: 0x0000000111e3bf02 _gi.cpython-37m-darwin.so`pygi_invoke_c_callable + 2146
        frame #44: 0x0000000111e3cdbd _gi.cpython-37m-darwin.so`pygi_function_cache_invoke + 61
        frame #45: 0x00000001000e8bbf Python`PyObject_Call + 136
        frame #46: 0x0000000100176f43 Python`_PyEval_EvalFrameDefault + 7144
        frame #47: 0x000000010017e7a6 Python`_PyEval_EvalCodeWithName + 1870
        frame #48: 0x00000001000e8a77 Python`_PyFunction_FastCallKeywords + 225
        frame #49: 0x000000010017def7 Python`call_function + 753
        frame #50: 0x0000000100176c57 Python`_PyEval_EvalFrameDefault + 6396
        frame #51: 0x00000001000e8e8c Python`function_code_fastcall + 112
        frame #52: 0x000000010017def7 Python`call_function + 753
        frame #53: 0x0000000100176d0b Python`_PyEval_EvalFrameDefault + 6576
        frame #54: 0x00000001000e8e8c Python`function_code_fastcall + 112
        frame #55: 0x000000010017def7 Python`call_function + 753
        frame #56: 0x0000000100176d0b Python`_PyEval_EvalFrameDefault + 6576
        frame #57: 0x000000010017e7a6 Python`_PyEval_EvalCodeWithName + 1870
        frame #58: 0x00000001001752b8 Python`PyEval_EvalCode + 51
        frame #59: 0x00000001001a394b Python`run_mod + 54
        frame #60: 0x00000001001a2975 Python`PyRun_FileExFlags + 163
        frame #61: 0x00000001001a201b Python`PyRun_SimpleFileExFlags + 263
        frame #62: 0x00000001001ba89e Python`pymain_main + 5389
        frame #63: 0x00000001001baf80 Python`_Py_UnixMain + 56
        frame #64: 0x00007fff684923d5 libdyld.dylib`start + 1

eblot, 2019-08-13:
I'm unable to get a working Python interpreter on macOS - I lost over two hours fighting this, I give up. I would not have imagined it would be such as mess to get an executable with debug symbols on macOS...

LLDB cannot therefore use the interpreter symbols to provide a useful insight, and GDB makes Python to crash at startup, I do not know why.

eblot, 2019-08-14:
Ok, after quite some troubles with Python and Homebrew, I've eventually been able to install 
[cpython-lldb](https://github.com/malor/cpython-lldb/issues/13) and obtain information about the Python traceback when this crash occurs:

    Traceback (most recent call last):
      File "/usr/local/Cellar/gnuradio/3.8.0.0/libexec/bin/gnuradio-companion", line 102, in <module>
        run_main()
      File "/usr/local/Cellar/gnuradio/3.8.0.0/libexec/bin/gnuradio-companion", line 95, in run_main
        exit(main())
      File "/usr/local/Cellar/gnuradio/3.8.0.0/lib/python3.7/dist-packages/gnuradio/grc/main.py", line 88, in main
        sys.exit(app.run())
      File "/usr/local/lib/python3.7/site-packages/gi/overrides/Gio.py", line 44, in run
        return Gio.Application.run(self, *args, **kwargs)
      File "/usr/local/Cellar/gnuradio/3.8.0.0/lib/python3.7/dist-packages/gnuradio/grc/gui/DrawingArea.py", line 154, in _handle_mouse_button_release
        coordinate=self._translate_event_coords(event),
      File "/usr/local/Cellar/gnuradio/3.8.0.0/lib/python3.7/dist-packages/gnuradio/grc/gui/canvas/flowgraph.py", line 737, in handle_mouse_selector_release
        self._new_connection = None    

It matches the investigation and troubleshooting Jarl has already one on this issue.

eblot, 2019-08-14:
The same issue can be reproduced by clearing out the Cairo context:

    self._new_connection._current_cr = None

so it looks like it is related to how GRC/pycairo/cairo, is managing the contexts with a CoreGraphics backend.

eblot, 2019-08-14:
It seems to be related to how `Connection._current_cr` is managed.

When the draw context is not saved for delayed use from `what_is_selected()`, the crash disappears. Even more surprising,  #2736 does not pop up!

Could it be related to the lifetime of the context, *i.e.* is it legal to keep a reference on the draw context?

eblot, 2019-08-15:
It would be interesting to know whether the current implementation (which seems ~ 3 years old) is actually valid, as I am suspicious about the lifetime of the Cairo context: 

 *  *is it valid to store a reference to a context outside the `draw()` sequence?*

I think we (at least, I) would need some advice from the Cairo developers.

I written a PoC as an alternative way to test the line path without relying on the Cairo backend.
The crash no longer occurs using this sequence (and #2136 is also silenced): https://github.com/eblot/gnuradio/commit/61be64deb0cdfc6b356d4391151b1134ddee7437

It does not mean it is the way to go, it just try to avoid keeping a reference on a (Quartz-backed) context. There may be better alternatives to implement.

eblot, 2019-08-19:
It would be interesting to know whether the current implementation (which seems ~ 3 years old) is actually valid, as I am suspicious about the lifetime of the Cairo context: it is valid to store a reference to a context outside the `draw()` sequence? We would need some advice from the Cairo developers.

I written a PoC as an alternative way to test the line path without relying on the Cairo backend.
The crash no longer occurs with this sequence. It does not mean it is the way to go :-)

jnicolson, 2019-08-25:
@eblot I've been fighting to figure out whats happening, I can't help but feel it's a Cairo bug with the Quartz backend.  I'm pretty amateur at debugging (especially with C) but so far i'm still not sure initialising a quartz backed cairo surface does an initial Core Graphics context save (where as the destructor is obviously expecting one given that it does a restore without checking).  

After a lot of struggling I gave up and came to the same conclusion as you and replaced the quartz context with an ImageSurface.  I'm still frustrated that I don't know enough to understand Cairo well enough to figure out the actual reason for the crashes - All I got to is they only occurred when a connection was present and that it was related to the Core Graphics context state save within Cairo..

eblot, 2019-08-25:
Hi Jarl,

Thanks for the feedback.

I'm afraid that we need to get some advice from of the Cairo Quartz backend developer. It may not only be related to the Quartz backend, btw (*i.e.*  it's not because the current implementation works with non-Quartz backend that it is indeed valid). It is valid to keep a reference onto the draw context forever and use it at anytime?

eblot, 2019-08-30:
There is a remaining issue: as soon as a connection is defined, it disappears immediately.
One needs to click on either port of the connection for make the connection appears. Very weird.

jnicolson, 2019-09-03:
@eblot If i'm understanding your last comment comment correctly I think this is the same on other backends (e.g. running on Ubuntu i get the same).  

It feels like there's room for improvement in the UX for connections - dragging from an Out port to an In port shows the connection dragging as expected but doesn't make a connection, single clicking on an Out port flashes the connection, but then it disappears but it obviously sets some state as when you click the In port the connection is made and finally selecting a connection after it's established then dragging it just drops the connection (if dragging it is not a valid action it feels like it should do nothing rather than drop it).

However this was all on Linux so nothing to do with the Quartz backend.  I still haven't made an progress understanding the crashes - have you managed to reach out to a Cairo Quartz developer at all?

eblot, 2019-09-03:
> @eblot If i'm understanding your last comment comment correctly I think this is the same on other backends (e.g. running on Ubuntu i get the same).

@jnicolson  Wow, ok. I did not even checked on Linux, as I would have imagined GR 3.8 final would not have been released with this kind of issue on what seems the main target platform for GR (Linux/Ubuntu) :-)

> Have you managed to reach out to a Cairo Quartz developer at all?

Actually I have not tried, as I did imagine that a core GR developer would better be in position to contact Cairo team than a mere user. Is anyone in charge? I can try to contact them, but I guess it is better if only one people post to their ML/ticket system.

eblot, 2019-09-03:
BTW, there are indeed many usuability issues with connections. 

Have you tried to add a trace to the connection's `draw()` method? It is called so many times for now no obvious reason there might be other deeper/hidden issue: once the port block is clicked to make it appear as black color, the connection is redraw ... 180+ times without changing anything on the DrawingArea (no move, no resize, no invalidation)! This is quite crazy.

I also (re)discover a side effect: raising an exception in the connection's `draw()` method always triggers the zoom 2:1 issue I was observing before the workaround to avoid storing the context :-)

iRonJ, 2020-03-29:
> The crash occurs when an attempt to connect an existing block is made by clicking/releasing a block inlet/outlet.
>
>
> This one addresses the drag n' drop issue. Drag n' drop issue prevents creating [...]

I implemented this change on a local copy of the 3.8 branch, recompiled for Mac OS Catalina (10.15.3)

I get a crash this error when trying to connect two blocks:
`Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/gnuradio/grc/gui/Application.py", line 327, in _handle_action
    top_block  = flow_graph.get_block("top_block")
  File "/usr/local/lib/python3.7/site-packages/gnuradio/grc/core/FlowGraph.py", line 171, in get_block
    raise KeyError('No block with name {!r}'.format(name))
KeyError: "No block with name 'top_block'"
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/gnuradio/grc/gui/Application.py", line 327, in _handle_action
    top_block  = flow_graph.get_block("top_block")
  File "/usr/local/lib/python3.7/site-packages/gnuradio/grc/core/FlowGraph.py", line 171, in get_block
    raise KeyError('No block with name {!r}'.format(name))
KeyError: "No block with name 'top_block'"
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/gnuradio/grc/gui/Application.py", line 327, in _handle_action
    top_block  = flow_graph.get_block("top_block")
  File "/usr/local/lib/python3.7/site-packages/gnuradio/grc/core/FlowGraph.py", line 171, in get_block
    raise KeyError('No block with name {!r}'.format(name))
KeyError: "No block with name 'top_block'"
Assertion failed: (!x_link_empty_p(&s->stack)), function CGGStackRestore, file /BuildRoot/Library/Caches/com.apple.xbs/Sources/CoreGraphics/CoreGraphics-1348.16/CoreGraphics/Context/CGGStack.c, line 74.
`

If I open a file too that has blocks not visible in the current window view, the entire GUI gets scaled to a much larger size, and mouse clicks are not aligned with what it looks like you're clicking on.

nickoe, 2020-03-30:
@iRonJ  please use more backticks for multiline code.

thinkski, 2020-08-31:
Ran into the same issue in 3.8.2.0 (via Homebrew) on MacOS 10.14.6. Did get this warning while installing:
```
Warning: gnuradio dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
```
So attempted to reinstall gcc, building from source, as well as gtk+3 and cairo. Received this warning while reinstalling gtk+3 and cairo:
```
Warning: Your Xcode (10.3) is outdated.
Please update to Xcode 11.3.1 (or delete it).
Xcode can be updated from the App Store.
```
So upgraded Xcode to 11.3.1 and rebuilt again. That fixed the warnings, but still GRC crashes the instance mouse button is released while drawing a connection from a block:
```
$ GDK_SCALE=2 gnuradio-companion
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.8.2.0 >>>

Block paths:
	/usr/local/Cellar/gnuradio/3.8.2.0/share/gnuradio/grc/blocks

Loading: "demo.grc"
>>> Done
Assertion failed: (s->stack->next != NULL), function CGGStackRestore, file /BuildRoot/Library/Caches/com.apple.xbs/Sources/CoreGraphics/CoreGraphics-1265.11/CoreGraphics/Context/CGGStack.c, line 77.
Abort trap: 6
```

ra1nb0w, 2020-08-31:
@thinkski have you tried the macports version?

thinkski, 2020-08-31:
> @thinkski have you tried the macports version?

I have not. Does the macports version work for you? If so, I'd be curious to investigate what it does differently -- that would suggest the issue is due to dependency version differences as I suspect it compiles the source in a similar way.

ra1nb0w, 2020-08-31:
> I have not. Does the macports version work for you? 

yes

rgov, 2020-12-04:
Are there clear repro steps for this? After some light testing I did not see a crash making a connection between two blocks.

I am using GNU Radio from Homebrew plus a rebuilt Cairo from [this branch](https://gitlab.freedesktop.org/jralls/cairo/-/tree/bigsur-use-after-free) which makes some changes to the Quartz backend to fix an issue exacerbated by Big Sur. It seems to be working for me, but I don't know if it is relevant to the issue at hand.

ra1nb0w, 2020-12-04:
@rgov those patches will be merged to upstream? I am still using this patch on macports and it works fine.

eblot, 2020-12-04:
> Are there clear repro steps for this? After some light testing I did not see a crash making a connection between two blocks.

Sorry, I gave up using GNU radio one year ago (too many issues at too many levels), so I'm afraid I cannot help here anymore.

rgov, 2020-12-04:
> @rgov those [Cairo] patches will be merged to upstream? I am still using this patch on macports and it works fine.

I would assume so. If they pass code review and seem to resolve the issue, I don't know why they would not be. The author of the patches participates in Cairo development, as far as I can tell.

ra1nb0w, 2020-12-05:
Ok. When released I will try to check if this patch is still needed.

NeuroForLunch, 2021-06-24:
Update on this? Or does #4814 fix it?

ra1nb0w, 2021-06-25:
In macports I am using the two following patches for GR 3.8:
- https://github.com/macports/macports-ports/blob/master/science/gnuradio/files/patch-quartz_drag_n_drop.38.diff
- https://github.com/macports/macports-ports/blob/master/science/gnuradio/files/patch-quartz_drop_crash.38.diff

dkozel, 2021-06-26:
Patch 1 was merged in https://github.com/gnuradio/gnuradio/pull/3980 for 3.9+. I've just flagged it for backporting to 3.8.
The second patch seems platform agnostic and if it's fixing a crash then should be merged? I've asked @ryanvolz if it's included in the Conda build for Windows where I've seen a similar crash.

ra1nb0w, 2021-06-26:
thank you @dkozel
if you need more information feel free to ask.

dkozel, 2021-06-27:
@ra1nb0w @ryanvolz Patch 1 is now backported to maint-3.8.
https://github.com/gnuradio/gnuradio/pull/4819

ra1nb0w, 2021-06-27:
thank you @dkozel

ra1nb0w, 2022-02-11:
just a note: 3.10.1.1 still has the quartz issue when you link blocks (patch 2)
```
Assertion failed: (s->stack->next != NULL), function CGGStackRestore, file /BuildRoot/Library/Caches/com.apple.xbs/Sources/CoreGraphics/CoreGraphics-1265.13/CoreGraphics/Context/CGGStack.c, line 77.
```

closed 2022-11-17 by mormj