gnuradio/gnuradio #2727: GRC38 OSX +Quartz: Unable to drag n' drop blocks

eblot, 2019-08-12:
Dropping block in drawing area does nothing when GRC 3.8 is used on macOS +quartz.

When the block selected from the tree view is dropped onto the drawing area, it simply disappears. This seems to come from a configuration issue with string drag n' drop.

Proposed fix is [eblot@62f3065](https://github.com/eblot/gnuradio/commit/62f3065f546813cab754b10b270c8a7af7664a63), after initial proposal and feedback from [GTK discourse thread](https://discourse.gnome.org/t/question-related-to-setting-up-text-drag-and-drop-pygtk3/1542/5).

eblot, 2019-08-13:
Simpler fix: 5ee92ee7

michaelld, 2019-08-13:
@eblot Wow that's a -very- simple patch! Are you & Jarl 100% sure this is what's needed? I'll test on my +x11 install to verify it does no harm there ...

eblot, 2019-08-13:
Well, at least, it fixes the initial issue with drag n' drop on my machine.

The only point is the ID I chose (`1`): it seems not to be used w/ GRC, and I did not take a chance to redefine it as `0` *i.e.* the same as `STRING`, as I do not know whether it could cause trouble to the GTK lib to use ids that would not be unique.

jnicolson, 2019-08-13:
I don't think this should cause any negative impacts on other platforms as it's purely an additional target (though i'd be interested why a drop target of UTF8_STRING is automatically selected on MacOS).  

I do think right now the code is slightly ambiguous - on the source side the target is defined by the DND_TARGETS array, but on the dest side it's defined by the drag_dest_add_text_targets() (which includes STRING and UTF8_STRING according to https://github.com/GNOME/gtk/blob/55839dc508acc1d646701c3c11cacd8b15e06c46/gtk/gtkselection.c#L168-L172). For a reason I don't understand though, trying to use the DND_TARGETS (which are scoped to same application vs drag_dest_add_text_targets() which seems to be scoped to any drag event) on the drag destination doesn't work, but this isn't a regression as it thats how things stand today..

dkozel, 2021-06-26:
Merged in 3980.

closed 2021-06-26 by dkozel