gnuradio/gnuradio #2730: Improve exception safety with smart pointers

elfring, 2019-08-13:
Would you like to wrap any pointer data members with the class template “[std::unique_ptr](https://en.wikipedia.org/wiki/Smart_pointer#unique_ptr "Description for the usage of smart pointers")”?

Update candidates:
- [dc_blocker_cc_impl](https://github.com/gnuradio/gnuradio/blob/d24b0c8f1ebe1a9681271c014659bf2ecc613e50/gr-filter/lib/dc_blocker_cc_impl.cc#L62)
- [random](https://github.com/gnuradio/gnuradio/blob/d24b0c8f1ebe1a9681271c014659bf2ecc613e50/gnuradio-runtime/lib/math/random.cc#L49)

marcusmueller, 2019-08-15:
Hm, you mean so the constructor doesn't leak memory in case it throws an exception?

I'd argue that in the case of these two classes, there's really no good reason the thigs `new`ed in the c'tor should actually be pointer members, or that they should be allocated manually in the constructor (instead of in the initializer list (pre-C++11, i.e. maint-3.7), or as class members (post-C++11, i.e. master + maint3.8).

But this is a good find! How did you stumble across these?

elfring, 2019-08-15:
I am trying to increase the development attention for specific software transformation possibilities.
How do you think about to take [the C++ guideline “R.11: Avoid calling new and delete explicitly”](https://isocpp.github.io/CppCoreGuidelines/CppCoreGuidelines#r11-avoid-calling-new-and-delete-explicitly "Recommendations around allocation and deallocation") into account?

marcusmueller, 2019-08-15:
I do like the idea of avoiding error-prone manual handling wherever possible, but the suggested fix (warn on every `new`, use `unique_ptr`) is a bit simplistic, and as said, not the right fix. The right fix would've been using object members instead of pointer members and have the C++ language take care of lifetimes, instead of delegating that to letting it deal with the lifetime of a `unique_ptr` and then letting that, being referenced *only* within the class itself, deal with object lifetime, which has the same effect, but is just slower (due to refcount or indirection) and confusing. 

That's why I asked you how you did find these two examples of bad `new` usage: When fixing a class of mistakes, we'd like to make sure we understand what we're doing, document it and do it consistently across the tree – we're a relatively large project.

So: what's the tooling that led you to find these two constructors?

elfring, 2019-08-15:
Such update candidates can be found also by [using the GitHub source code search interface](https://help.github.com/en/articles/searching-code "Searching code").
Are you looking for related development tools?

marcusmueller, 2019-08-15:
so you just searched for "new" in C++ files?!

elfring, 2019-08-15:
The C++ key word “`delete`” is easier (or safer) to find for the reconsideration of source code places, isn't it?

marcusmueller, 2019-08-15:
I asked what _you_ did.

elfring, 2019-08-15:
I tend to use the key word “`delete`” for some source code searches.

marcusmueller, 2019-08-15:
I don't want to sound rude (I really don't!), but I'm trying to figure out what specifically your search term was to find these specific two cases, not what you *tend* to do. Searching for `delete` as word in C++ source code in the GNU Radio tree yields 169 results.

elfring, 2019-08-15:
The presented search results can be filtered further by manual review.
Higher level tools can support semantic source code searches so that questionable places can be found in more convenient ways, can't they?

marcusmueller, 2019-08-15:
OK, I have work to do. I asked you several times now what **you** specifically did, it doesn't seem to get any less vague, so I'm going to close this issue since the instance you pointed at is problematic, and needs to be fixed, but not in the proposed way, and you show no inclination of sharing the methodology you use with us, so that we can apply it consistently within our own coding standards.

closed 2019-08-15 by marcusmueller

elfring, 2019-08-15:
I hope that a better common understanding can still be achieved also for this issue.
Would you like to [improve the affected software any further](https://github.com/gnuradio/gnuradio/search?l=C%2B%2B&q=delete "Searching for update candidates")?

marcusmueller, 2019-08-15:
thank you very much.

elfring, 2019-08-15:
It might become possible to generate patches automatically to fix remaining open issues.

marcusmueller, 2019-08-15:
Thank you for your outstanding contributions on the field of vagueness. We will no further consider this issue.

elfring, 2019-08-15:
Under which circumstances would you find the corresponding need for change not so vague any more?