gnuradio/gnuradio #2731: Can't build master commit d50e7b504e

Antanide, 2019-08-14:
This is what I get while building.
```
[ 85%] Generating moc_display_qt.cpp
moc: Cannot create /home/fabio/tsm/linux-tools/gnuradio/build/gr-qtgui/examples/c++/moc_display_qt.cpp
make[2]: *** [gr-qtgui/examples/c++/CMakeFiles/display_qt.dir/build.make:62: gr-qtgui/examples/c++/moc_display_qt.cpp] Error 1
make[1]: *** [CMakeFiles/Makefile2:10955: gr-qtgui/examples/c++/CMakeFiles/display_qt.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```
building version  v3.8.0.0-rc2  worked fine.

Antanide, 2019-08-14:
Worked around the issue by running "make" with superuser privileges.

closed 2019-08-14 by Antanide

marcusmueller, 2019-08-17:
running make as superuser is **always** wrong. Don't do it. You have a different problem, and it's most probably because you've run the last make as root. Don't do that!

Delete your build directory (you'll need superuser privileges for that now, because the files belong to root), and then build from scratch – but not as root.

Anyway, this is not a GNU Radio bug, but expected behaviour from your file system: It denies a normal user the right to change a file created by root.