gnuradio/gnuradio #2736: GRC38 OSX +Quartz refresh window content with 2:1 zoom on block port selection

eblot, 2019-08-14:
When one input or output block is selected in GRC, the drawing area is refreshed with a 2:1 copy of the actual content (100% reproducible). See attached screenshot:

[image]

Another symptom is a 2:1 full window redraw, including the menu bar & icon bar, although it is less likely.

[image]

eblot, 2019-08-15:
It looks like another symptom of #2726. Whenever #2726 is worked around, this symptom disappears as well.

iracigt, 2019-08-24:
I also encountered this issue, but only while running in retina (HiDPI) mode. I was able to reliably work around it by launching GRC from an app bundle with "Open in Low Resolution" set.

jwanga, 2020-02-12:
> I also encountered this issue, but only while running in retina (HiDPI) mode. I was able to reliably work around it by launching GRC from an app bundle with "Open in Low Resolution" set.

Thanks for this advise. How did you create a GRC app bundle?

rgov, 2020-12-04:
I am not seeing anything suddenly resize when I click on a block under Big Sur. Does this issue still repro?

If anything, the canvas area is probably half the size it should be, because the text is a bit difficult to read.

NeuroForLunch, 2021-06-24:
> I am not seeing anything suddenly resize when I click on a block under Big Sur. Does this issue still repro?
> 
> If anything, the canvas area is probably half the size it should be, because the text is a bit difficult to read.

I can't reproduce this issue and I think #4814 fixes this

michaelld, 2021-09-25:
Yes I think the generic fix in #4814 addresses this issue. I'm going to go ahead and close; please reopen if it's still an issue & post more info about what's going on.

closed 2021-09-25 by michaelld

m-anish, 2021-12-28:
FWIW. I'm seeing something like this happen with a recent install #5429