gnuradio/gnuradio #2745: Could NOT find MPLIB with gr-osmosdr

canghaiwuhen, 2019-08-18:
Can't install gr-osmosdr，said that

-- Checking for module 'gmp'
--   No package 'gmp' found
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_INCLUDE_DIR)

ferrellsl, 2019-08-23:
I can confirm this problem on my system as well.

velichkov, 2019-08-28:
Hi @canghaiwuhen,

For the `MPLIB` error you need to install `libgmp-dev` package.

Note that gr-osmosdr still [does not support gnuradio 3.8](https://osmocom.org/issues/3855) but [here is a fork](https://github.com/igorauad/gr-osmosdr/tree/gr3.8) that contains some patches. For `pybombs` you'll need to update the source url in `gr-osmosdr.lwr` receipt.

dkozel, 2019-08-29:
Hi @canghaiwuhen, thanks for reporting this. Velichkov is correct that gr-osmosdr has not been updated to support GNU Radio 3.8 yet, so PyBOMBS' error is at least partially correct.

This issue is best posted on the PyBOMBS recipes repository since thats where a recipe fix would go, or to the gr-osmosdr repository/mailing list as that's where a fix is likely to come from.
https://github.com/gnuradio/gr-recipes

closed 2019-08-29 by dkozel

anthropic-radio-source, 2019-08-29:
Can someone update gr-iqbal. It does not install with gnuradio 3.8 or 3.9. I have been banging my head against the wall trying to install it all week to no avail. Anyone know a work around?

velichkov, 2019-08-29:
Hi @anthropic-radio-source,

Try `gr3.8` branch from [this fork](https://github.com/mvaenskae/gr-iqbal/tree/gr3.8).

anthropic-radio-source, 2019-08-29:
Vasil

 

I’ll give this fork a go - again!

 

Regards

 

Alistair

canghaiwuhen, 2019-09-01:
> Hi @canghaiwuhen,
>
> For the `MPLIB` error you need to install `libgmp-dev` package.
>
> Note that gr-osmosdr still [does not support gnuradio 3.8](https://osmocom.org/issues/3855) but [here is a [...]

I changed it and it working!
Thank you very much!

anthropic-radio-source, 2019-09-01:
I have libgmp-dev installed (by pybombs). I have tried the fork you suggested and it would not install and comes up with the same error – no gmp or mplib installed. No idea what is causing this. I scrubbed the hard disk yesterday and started with a fresh copy of ubuntu 18.04. still get the same problem. UHD/Gnuradio installs OK with pybombs but neither the igorauad or the mvaeaskae forks of gr-iqbal/gr-osmosdr work for me. No idea why!

canghaiwuhen, 2019-09-01:
Maybe you need try again, I changed 2 lwr, and OK