gnuradio/gnuradio #2748: Rotator is broken in 3.8

argilo, 2019-08-19:
While porting one of my out-of-tree modules to GNU Radio 3.8, I noticed that the Rotator block is broken. The problem can be reproduced with the following flowgraph:

[image]

This produces the following output:

[image]

I haven't yet had time to investigate what's going on here.

drmpeg, 2019-08-20:
The rotator block works fine on ARM, so most likely it's a VOLK issue.

I haven't tested on X86 yet.

argilo, 2019-08-20:
That seems likely. I changed `#if 0` to `#if 1` here to prevent VOLK from being used, and the rotator went back to working correctly:

https://github.com/gnuradio/gnuradio/blob/39c922e8b28eb67515c134712a4bfb0bed3cd1f7/gr-blocks/lib/rotator_cc_impl.cc#L55-L70

argilo, 2019-08-20:
I suspect https://github.com/gnuradio/volk/commit/05b92b79c17b80a08c564a73f4386a573f60a610 introduced the bug; when I comment out the rotator FMA protokernels from that commit, the rotator starts working properly again.

argilo, 2019-08-20:
I tracked down the bug. A fix is here: https://github.com/gnuradio/volk/pull/278

I'm curious why this wasn't caught by tests, since qa_rotator.cc appears to test `rotateN`.

argilo, 2019-08-20:
When I run `build/gr-blocks/lib/blocks_qa_rotator.cc` on its own, it fails:
```
Running 2 test cases...
/home/argilo/prefix_next/src/gnuradio38/gr-blocks/lib/qa_rotator.cc(95): error: in "t2": check std::abs(expected - actual) <= 0.0001 has failed
/home/argilo/prefix_next/src/gnuradio38/gr-blocks/lib/qa_rotator.cc(95): error: in "t2": check std::abs(expected - actual) <= 0.0001 has failed
/home/argilo/prefix_next/src/gnuradio38/gr-blocks/lib/qa_rotator.cc(95): error: in "t2": check std::abs(expected - actual) <= 0.0001 has failed
...
/home/argilo/prefix_next/src/gnuradio38/gr-blocks/lib/qa_rotator.cc(95): error: in "t2": check std::abs(expected - actual) <= 0.0001 has failed
/home/argilo/prefix_next/src/gnuradio38/gr-blocks/lib/qa_rotator.cc(95): error: in "t2": check std::abs(expected - actual) <= 0.0001 has failed
/home/argilo/prefix_next/src/gnuradio38/gr-blocks/lib/qa_rotator.cc(95): error: in "t2": check std::abs(expected - actual) <= 0.0001 has failed

*** 1692 failures are detected in the test module "Master Test Suite"
```
But `make test` passes:
```
        Start 158: blocks_qa_rotator.cc
158/364 Test #158: blocks_qa_rotator.cc ...............................   Passed    0.16 sec
```
:confused:

argilo, 2019-08-20:
Aha, it's because `build/gr-blocks/lib/blocks_qa_rotator.cc_test.sh` runs the test with `export VOLK_GENERIC=1`, which I assume prevents the buggy protokernel from running.

argilo, 2019-10-17:
It's been two months now and several people have encountered this bug. Progress on volk seems to have stalled, and so the fix has not been merged there.

michaelld, 2019-10-17:
Sorry that took so long! Hopefully we'll do a better job in the future. Busy-ness at work happens!

argilo, 2019-10-17:
Thanks for getting the volk fix merged! I think this report should stay open for the time being though, since GNU Radio still needs to pick up the fix on the master and maint-3.8 branches.

michaelld, 2019-10-17:
Acknowledged & you are correct. I thought this issue was for Volk, not GR ...

argilo, 2019-11-09:
The rotator fixes were pulled into GNU Radio's master branch in abba02bd39b6f6fabb63755b4909d1d3f7d8ba4b, but the maint-3.8 branch is still broken. According to @mbr0wn the changes will need to go into a volk release first.

michaelld, 2019-11-10:
That's true. We can allow master to use any arbitrary commit. We need the maint branches to be using a release, for stability sake ... that's where we do releases of GR from, and GR must use a Volk release. We're working on getting a new 2.0.1 Volk release out ... "we" meaning Marcus showing us how to do it & then hopefully we'll get more regular releases :)

argilo, 2019-11-10:
Sounds good. I'm glad to hear a volk release is on the way.

devnulling, 2020-01-05:
@marcusmueller @michaelld @argilo Does the main-3.8 branch need to be updated for the new Volk 2.1.0 pointer https://github.com/gnuradio/volk/tree/v2.1.0 ? or can this issue be closed out with https://github.com/gnuradio/gnuradio/pull/2853 (it looks like it was merged into maint-3.8) ?

argilo, 2020-01-05:
#2853 was only merged into master. The maint-3.8 branch still uses VOLK 2.0.0 and needs to pick up the fix.

argilo, 2020-01-16:
This was fixed on the maint-3.8 branch in https://github.com/gnuradio/gnuradio/pull/3062, which bypasses VOLK in the rotator block. I'll do a quick test to confirm the rotator works and then close this off.

argilo, 2020-01-16:
I confirmed that the rotator block works properly on the latest maint-3.8 (commit cdbe702b484aa9a88613aef11d23cfc9aea21416).

closed 2020-01-16 by argilo