gnuradio/gnuradio #2751: qtgui: time sink with multiple ports as variable error

mormj, 2019-08-20:
When using QT gui time sink with multiple inputs, but having the Num Inputs parameter specified as a variable creates a generate error:

Simple flowgraph 
[image]

works if Number of Inputs is set to 2, but when set to the variable num_ports (which is 2) the following error occurs

`
Generating: '/share/git/gr-aep/build/test_qtgui_ports_variable.py'
Generate Error: (ValueError("invalid literal for int() with base 10: 'num_ports'",), '<%\n    win = \'self._%s_win\'%id\n%>\\\nqtgui.${type.fcn}(\n    ${size}, #size\n    ${srate}, #samp_rate\n    ${name}, #name\n    ${0 if type.startswith(\'msg\') else nconnections} #number of inputs\n)\nself.${id}.set_update_time(${update_time})\nself.${id}.set_y_axis(${ymin}, ${ymax})\n\nself.${id}.set_y_label(${ylabel}, ${yunit})\n\nself.${id}.enable_tags(${entags})\nself.${id}.set_trigger_mode(${tr_mode}, ${tr_slope}, ${tr_level}, ${tr_delay}, ${tr_chan}, ${tr_tag})\nself.${id}.enable_autoscale(${autoscale})\nself.${id}.enable_grid(${grid})\nself.${id}.enable_axis_labels(${axislabels})\nself.${id}.enable_control_panel(${ctrlpanel})\nself.${id}.enable_stem_plot(${stemplot})\n\n% if legend == "False":\nself.${id}.disable_legend()\n% endif\n\nlabels = [${label1}, ${label2}, ${label3}, ${label4}, ${label5},\n    ${label6}, ${label7}, ${label8}, ${label9}, ${label10}]\nwidths = [${width1}, ${width2}, ${width3}, ${width4}, ${width5},\n    ${width6}, ${width7}, ${width8}, ${width9}, ${width10}]\ncolors = [\'${color1}\', \'${color2}\', \'${color3}\', \'${color4}\', \'${color5}\',\n    \'${color6}\', \'${color7}\', \'${color8}\', \'${color9}\', \'${color10}\']\nalphas = [${alpha1}, ${alpha2}, ${alpha3}, ${alpha4}, ${alpha5},\n    ${alpha6}, ${alpha7}, ${alpha8}, ${alpha9}, ${alpha10}]\nstyles = [${style1}, ${style2}, ${style3}, ${style4}, ${style5},\n    ${style6}, ${style7}, ${style8}, ${style9}, ${style10}]\nmarkers = [${marker1}, ${marker2}, ${marker3}, ${marker4}, ${marker5},\n    ${marker6}, ${marker7}, ${marker8}, ${marker9}, ${marker10}]\n\n\n% if type.endswith(\'complex\'):\nfor i in range(${2 if type.startswith(\'msg\') else 2*int(nconnections)}):\n    if len(labels[i]) == 0:\n        if (i % 2 == 0):\n            self.${id}.set_line_label(i, "Re{{Data {0}}}".format(i/2))\n        else:\n            self.${id}.set_line_label(i, "Im{{Data {0}}}".format(i/2))\n    else:\n        self.${id}.set_line_label(i, labels[i])\n    self.${id}.set_line_width(i, widths[i])\n    self.${id}.set_line_color(i, colors[i])\n    self.${id}.set_line_style(i, styles[i])\n    self.${id}.set_line_marker(i, markers[i])\n    self.${id}.set_line_alpha(i, alphas[i])\n% else:\nfor i in range(${1 if type.startswith(\'msg\') else int(nconnections)}):\n    if len(labels[i]) == 0:\n        self.${id}.set_line_label(i, "Data {0}".format(i))\n    else:\n        self.${id}.set_line_label(i, labels[i])\n    self.${id}.set_line_width(i, widths[i])\n    self.${id}.set_line_color(i, colors[i])\n    self.${id}.set_line_style(i, styles[i])\n    self.${id}.set_line_marker(i, markers[i])\n    self.${id}.set_line_alpha(i, alphas[i])\n% endif\n\n${win} = sip.wrapinstance(self.${id}.pyqwidget(), Qt.QWidget)\n${gui_hint() % win}')
>>> Failure
`

This is on master/maint-3.8 branch - 39c922e8b28eb67515c134712a4bfb0bed3cd1f7

cbdonohue, 2019-10-03:
It appears that in one of the Mako substitutions it is trying to perform an int() on what could be an integer number of connections or a string (num_ports) that represents the number of connections (from a variable block).

Relevant code:
https://github.com/gnuradio/gnuradio/blob/55c55fe3e17d556484ed9139e8e941a0da00f9c4/gr-qtgui/grc/qtgui_time_sink_x.block.yml#L1053

https://github.com/gnuradio/gnuradio/blob/55c55fe3e17d556484ed9139e8e941a0da00f9c4/gr-qtgui/grc/qtgui_time_sink_x.block.yml#L1067

willcode, 2021-03-03:
Yes, it is trying to execute `int('num_ports')`.

marcusmueller, 2024-01-07:
Can still reproduce, fails to compile to Python.. Example fg: https://gist.github.com/marcusmueller/d801e9e0b29158cf046b3547e01f76e8