gnuradio/gnuradio #2753: ImportError: No module named gnu-radio

isaactd92, 2019-08-22:
I'm trying to create out of tree modules in gnu-radio follow the next instructions:
https://wiki.gnuradio.org/index.php/OutOfTreeModules
When I finally make test appears a problem: 
(base) itupac@itupac:~/workarea-gnuradio/gr-howto/build$ make test
Running tests...
Test project /home/itupac/workarea-gnuradio/gr-howto/build
    Start 1: test_howto
1/2 Test #1: test_howto .......................   Passed    0.00 sec
    Start 2: qa_square_ff
2/2 Test #2: qa_square_ff .....................***Failed    0.02 sec

50% tests passed, 1 tests failed out of 2

Total Test time (real) =   0.03 sec

The following tests FAILED:
	  2 - qa_square_ff (Failed)
Errors while running CTest
Makefile:85: recipe for target 'test' failed
make: *** [test] Error 8

I put ctest -V to descover it and appears:
(base) itupac@itupac:~/workarea-gnuradio/gr-howto/build$ ctest -V
UpdateCTestConfiguration  from :/home/itupac/workarea-gnuradio/gr-howto/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/itupac/workarea-gnuradio/gr-howto/build/DartConfiguration.tcl
Test project /home/itupac/workarea-gnuradio/gr-howto/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 1
    Start 1: test_howto

1: Test command: /bin/sh "/home/itupac/workarea-gnuradio/gr-howto/build/lib/test_howto_test.sh"
1: Test timeout computed to be: 9.99988e+06
1: 
1: 
1/2 Test #1: test_howto .......................   Passed    0.00 sec
test 2
    Start 2: qa_square_ff

2: Test command: /bin/sh "/home/itupac/workarea-gnuradio/gr-howto/build/python/qa_square_ff_test.sh"
2: Test timeout computed to be: 9.99988e+06
2: Traceback (most recent call last):
2:   File "/home/itupac/workarea-gnuradio/gr-howto/python/qa_square_ff.py", line 22, in <module>
2:     from gnuradio import gr, gr_unittest
2: ImportError: No module named gnuradio
2/2 Test #2: qa_square_ff .....................***Failed    0.02 sec

50% tests passed, 1 tests failed out of 2

Total Test time (real) =   0.03 sec

The following tests FAILED:
	  2 - qa_square_ff (Failed)
Errors while running CTest

I'm skipping or doing something wrong but I'm not sure in what part
I'll appreciate your help
Thaks

michaelld, 2019-08-22:
which version of GR are you using?

isaactd92, 2019-08-22:
It's 3.7.13.4

michaelld, 2019-08-22:
OK. What OS & version? Did you build GR from source, or use a package manager?

isaactd92, 2019-08-22:
Ubuntu 18.04.3 LTS
I have both now. From Source and from PyBombs.

michaelld, 2019-08-22:
OK. Can you post the file in the top-level build directory called `CMakeCache.txt`?

isaactd92, 2019-08-22:
# This is the CMakeCache file.
# For build in directory: /home/itupac/workarea-gnuradio/gr-howto/build
# It was generated by CMake: /usr/bin/cmake
# You can edit this file to change values found and used by cmake.
# If you do not want to change any of the values, simply exit the editor.
# If you do want to change a value, simply edit, save, and exit the editor.
# The syntax for the file is as follows:
# KEY:TYPE=VALUE
# KEY is the name of a variable in the cache.
# TYPE is a hint to GUIs for the type of VALUE, DO NOT EDIT TYPE!.
# VALUE is the current value for the KEY.

########################
# EXTERNAL cache entries
########################

//The directory containing a CMake configuration file for Boost.
Boost_DIR:PATH=Boost_DIR-NOTFOUND

//Boost filesystem library (debug)
Boost_FILESYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Boost filesystem library (release)
Boost_FILESYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_filesystem.so

//Path to a file.
Boost_INCLUDE_DIR:PATH=/usr/include

//Boost library directory DEBUG
Boost_LIBRARY_DIR_DEBUG:PATH=/usr/lib/x86_64-linux-gnu

//Boost library directory RELEASE
Boost_LIBRARY_DIR_RELEASE:PATH=/usr/lib/x86_64-linux-gnu

//Boost system library (debug)
Boost_SYSTEM_LIBRARY_DEBUG:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Boost system library (release)
Boost_SYSTEM_LIBRARY_RELEASE:FILEPATH=/usr/lib/x86_64-linux-gnu/libboost_system.so

//Path to a program.
CMAKE_AR:FILEPATH=/usr/bin/ar

//Choose the type of build, options are: None(CMAKE_CXX_FLAGS or
// CMAKE_C_FLAGS used) Debug Release RelWithDebInfo MinSizeRel.
CMAKE_BUILD_TYPE:STRING=

//Enable/Disable color output during build.
CMAKE_COLOR_MAKEFILE:BOOL=ON

//CXX compiler
CMAKE_CXX_COMPILER:FILEPATH=/usr/bin/c++

//A wrapper around 'ar' adding the appropriate '--plugin' option
// for the GCC compiler
CMAKE_CXX_COMPILER_AR:FILEPATH=/usr/bin/gcc-ar-7

//A wrapper around 'ranlib' adding the appropriate '--plugin' option
// for the GCC compiler
CMAKE_CXX_COMPILER_RANLIB:FILEPATH=/usr/bin/gcc-ranlib-7

//Flags used by the compiler during all build types.
CMAKE_CXX_FLAGS:STRING=

//Flags used by the compiler during debug builds.
CMAKE_CXX_FLAGS_DEBUG:STRING=-g

//Flags used by the compiler during release builds for minimum
// size.
CMAKE_CXX_FLAGS_MINSIZEREL:STRING=-Os -DNDEBUG
[... 494 lines cut ...]
//ADVANCED property for variable: SWIG_DIR
SWIG_DIR-ADVANCED:INTERNAL=1
//ADVANCED property for variable: SWIG_EXECUTABLE
SWIG_EXECUTABLE-ADVANCED:INTERNAL=1
//ADVANCED property for variable: SWIG_VERSION
SWIG_VERSION-ADVANCED:INTERNAL=1
//Last used Boost_ADDITIONAL_VERSIONS value.
_Boost_ADDITIONAL_VERSIONS_LAST:INTERNAL=1.35.0;1.35;1.36.0;1.36;1.37.0;1.37;1.38.0;1.38;1.39.0;1.39;1.40.0;1.40;1.41.0;1.41;1.42.0;1.42;1.43.0;1.43;1.44.0;1.44;1.45.0;1.45;1.46.0;1.46;1.47.0;1.47;1.48.0;1.48;1.49.0;1.49;1.50.0;1.50;1.51.0;1.51;1.52.0;1.52;1.53.0;1.53;1.54.0;1.54;1.55.0;1.55;1.56.0;1.56;1.57.0;1.57;1.58.0;1.58;1.59.0;1.59;1.60.0;1.60;1.61.0;1.61;1.62.0;1.62;1.63.0;1.63;1.64.0;1.64;1.65.0;1.65;1.66.0;1.66;1.67.0;1.67;1.68.0;1.68;1.69.0;1.69
//Components requested for this build tree.
_Boost_COMPONENTS_SEARCHED:INTERNAL=filesystem;system
//Last used Boost_INCLUDE_DIR value.
_Boost_INCLUDE_DIR_LAST:INTERNAL=/usr/include
//Last used Boost_LIBRARY_DIR_DEBUG value.
_Boost_LIBRARY_DIR_DEBUG_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_LIBRARY_DIR_RELEASE value.
_Boost_LIBRARY_DIR_RELEASE_LAST:INTERNAL=/usr/lib/x86_64-linux-gnu
//Last used Boost_NAMESPACE value.
_Boost_NAMESPACE_LAST:INTERNAL=boost
//Last used Boost_USE_MULTITHREADED value.
_Boost_USE_MULTITHREADED_LAST:INTERNAL=TRUE
__pkg_config_arguments_PC_CPPUNIT:INTERNAL=cppunit
__pkg_config_arguments_PC_GNURADIO_RUNTIME:INTERNAL=gnuradio-runtime
__pkg_config_checked_PC_CPPUNIT:INTERNAL=1
__pkg_config_checked_PC_GNURADIO_RUNTIME:INTERNAL=1
prefix_result:INTERNAL=/usr/local/lib

michaelld, 2019-08-22:
OK so I find this odd:

```
//Path to a program.
PYTHON_EXECUTABLE:FILEPATH=/home/itupac/anaconda2/bin/python2
//Path to a file.
PYTHON_INCLUDE_DIR:PATH=/usr/include/python2.7
//Path to a library.
PYTHON_LIBRARY:FILEPATH=/usr/lib/x86_64-linux-gnu/libpython2.7.so
```

It looks like you have an "apt"-based install of Py27, but also one from source. Those are -not- guaranteed to be compatible; it's always better to have a single Py2 (or Py3) install ... CMake just isn't very smart yet about matching up the various PYTHON parts ... it's easy to get the EXECUTABLE of one install with the LIBRARY of another and the INCLUDE_DIR of yet another (if you have multiple Python installs).

I'd advise you to drop back to a single Python install.

michaelld, 2019-08-22:
It looks like the found version of GNU Radio is the one in `/usr/local/`, and CMake did a good job finding it ... very consistent there!

michaelld, 2019-08-22:
What does `echo $PYTHONPATH` return?

isaactd92, 2019-08-22:
Nothing! 

[image]

michaelld, 2019-08-22:
Interesting. OK, what does `which python` return?

michaelld, 2019-08-22:
and `which python2`?

isaactd92, 2019-08-22:
$ which python
/home/itupac/anaconda2/bin/python
$ which python2
/home/itupac/anaconda2/bin/python2
$ which python2.7
/home/itupac/anaconda2/bin/python2.7

michaelld, 2019-08-22:
`echo $PATH`?

isaactd92, 2019-08-22:
echo $PATH
/home/itupac/anaconda2/bin:/home/itupac/anaconda2/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin

michaelld, 2019-08-22:
And this:
```
python2.7 -c "import sys ; print sys.path"
```

isaactd92, 2019-08-26:
Found the solution:
1. First at all, I was running in the environment of Anaconda, so I disable anaconda with : conda deactivate
2. I remove the created OOT: gr-how , as It was created inside the file workarea -gnuradio.
3. I created again gr-how but out of the file workarea-gnuradio, following the steps of the link.
4. Passed.

sharad1126, 2020-04-26:
@michaelld can you please help me with the same issue here? I am not able to make it because of the gnuradio issue. 

```
mbp-2 build % make
[ 70%] Built target grgsm
[ 70%] Built target grgsm_swig_swig_doc
[ 72%] Built target _grgsm_swig_swig_tag
[ 75%] Built target grgsm_swig_swig_2d0df
[ 78%] Built target pygen_swig_aa10d
[ 81%] Built target _grgsm_swig
[ 83%] Built target pygen_python_d8811
[ 85%] Built target pygen_python_misc_utils_dcb9d
[ 86%] Built target pygen_python_receiver_fc70a
[ 88%] Built target pygen_python_demapping_4545a
[ 90%] Built target pygen_python_transmitter_0db58
[ 92%] Built target pygen_python_trx_ff002
[ 93%] Generating grgsm_livemon
Traceback (most recent call last):
  File "/usr/local/Cellar/gnuradio/3.7.13.4_14/libexec/bin/grcc", line 29, in <module>
    from gnuradio import gr
ImportError: No module named gnuradio
make[2]: *** [apps/grgsm_livemon] Error 1
make[1]: *** [apps/CMakeFiles/pygen_apps.dir/all] Error 2
make: *** [all] Error 2
```

```
mbp-2 build % which python
/usr/bin/python
```

```
python --version
Python 2.7.16
```

```
mbp-2 build % $PYTHONPATH
zsh: permission denied: /usr/local/lib/python2.7/site-packages
```

isaactd92, 2020-04-27:
I think this is a different kind of problem. Your program is not recognizing gnuradio programme. Could you create another issue and continue there?
My possible solutions are: 
1. Run a script of the examples that come inside of the file  workarea -gnuradio or create a diagram flow in GNU Radio companion to seeif GNU radio is well installed. (I think is a problem of gnuradio)
2. If it works, Create a simple programe OOT and see if it runs. As the square example. 
3. Be sure to create it OUTSIDE of the workarea -gnuradio file
3. If any of them works, continue in another issue and tag me to see your results