gnuradio/gnuradio #2757: QT GUI Sink doesn't maximize itself at startup

MattEttus, 2019-08-23:
The QT GUI sink starts out very small, a small fraction of the open window.  Simply resizing the window causes the GUI sink to become the correct size, and everything is good after that.

This only applies to the QT GUI Sink, not the QT Time Sink, QT Frequency Sink, etc.

cbdonohue, 2020-01-09:
I think this has to do with how the layouts are stacked in the mako template.
I'm experimenting with changing this line:
https://github.com/gnuradio/gnuradio/blob/65a8c6c526d05ce5bc060725d9338e81a4bbbc22/grc/core/generator/flow_graph.py.mako#L77
to this:
`self.top_layout = Qt.QVBoxLayout(self.top_scroll)`
This seems to take into account all of the widgets in the scroll area when sizing the dialog box.

marcusmueller, 2024-01-07:
can indeed confirm, window is smaller than necessary on default startup:

[image]

reference flowgraph: https://gist.github.com/marcusmueller/c196b7d8329ccedf999f373de7749d5e

Doom4535, 2026-08-19:
I am also encountering this, it's kinda annoying as it spams my logs with messages untill I resize the QT window.