gnuradio/gnuradio #2758: grc locks if you send too much stdout to its console window

MattEttus, 2019-08-23:
Even after the flowgraph is killed, the grc locks if you send too much stdout to its console window.  Sometimes it recovers after a few minutes, but sometimes you just need to kill it.

mormj, 2019-08-23:
Second this.  The problem seems much worse in 3.8 than in 3.7.

marcusmueller, 2019-08-23:
To give you a bit of background:

The issue is well-known.

The output buffer is a GTK text field, which has a DOM.

Appending things to that is slow, and occasionally leads to tree structure rebalancing.

The solution is probably to replace the whole thing with something else. There's an actual console widget for GTK, but it's Linux-only.

We could be writing our own Gtk widget in C, build and install, and then use it in GRC. yeah. Not likely.

We could be getting extremely tricky about handling the scrollbar ourselves an only present the text field only with the data it needs to *acutely* display; we'd still need to handle all that data in Python, which is unfortunate.

So, for now, I'd see this kind of a "unlikely to be fixed; don't overdo stdout in your GRC applications" (stdout isn't really useful for application purposes – the multithreaded nature of GNU Radio and the libc, Python output, C++ iostream buffering make the whole thing non-deterministic; I think it might be a sensible medium-term move to explicitly deprecate using stdout in a GNU Radio block alltogether, but we'd need to write a vastly superior logging framework first).

mormj, 2019-09-05:
Disabling the scroll lock on the grc console window, then looking at the stdout in the terminal seems to alleviate the locking up issues for me

marcusmueller, 2024-03-11:
low due to qt-grc

martinsah, 2025-02-03:
Can we pipe stdout somewhere else? I'm sitting here watching line by line scroll out for 10 minutes so I don't have to hard-kill gnuradio-companion.

dm1sh, 2025-05-21:
3.10.12 release crashes in GNOME Wayland environment with only `Gdk-Message: [time]: Error flushing display: Broken pipe` displayed on massive logging.
It might be a good idea to add an option to open a new console window for the output of the running flowgraph.

marcusmueller, 2025-05-21:
@dm1sh  that seems to be a completely different problem. Please don't add different issues to existing ones! Please open a new bug report for what you're seeing!