gnuradio/gnuradio #2763: grc: global_blocks_path prevents switching gr versions

mormj, 2019-08-26:
When going back and forth between GNU Radio versions (e.g. 3.7 and 3.8), GRC sometimes gets stuck looking for blocks for the other gr version.  GR 3.7 crashes when it gets stuck looking for the 3.8 blocks because of the format change.  

Here is the error
`
Traceback (most recent call last):
  File "/share/gnuradio/gr37/bin/gnuradio-companion", line 99, in <module>
    run_main()
  File "/share/gnuradio/gr37/bin/gnuradio-companion", line 92, in run_main
    exit(main())
  File "/share/gnuradio/gr37/lib/python2.7/dist-packages/gnuradio/grc/main.py", line 51, in main
    install_prefix=gr.prefix()
  File "/share/gnuradio/gr37/lib/python2.7/dist-packages/gnuradio/grc/gui/Platform.py", line 38, in __init__
    _Platform.__init__(self, *args, **kwargs)
  File "/share/gnuradio/gr37/lib/python2.7/dist-packages/gnuradio/grc/core/Platform.py", line 75, in __init__
    self.build_block_library()
  File "/share/gnuradio/gr37/lib/python2.7/dist-packages/gnuradio/grc/core/Platform.py", line 175, in build_block_library
    hide_bokeh_gui_options_if_not_installed(self.blocks['options'])
  File "/share/gnuradio/gr37/lib/python2.7/dist-packages/gnuradio/grc/core/utils/odict.py", line 35, in __getitem__
    return self._data[key]
KeyError: 'options'
`

Workaround:
- in ~/.gnuradio/grc.conf
- comment out global_blocks_path ...

Not sure exactly what the correct solution is, but I think it would be preferable instead of global options in one .conf file, for the conf file to be local to or tied to the pybombs installation.

schneider42, 2020-07-01:
Same happens for me the other way round:

(System installed) GRC 3.7 seem to have added `global_blocks_path` to `~/.gnuradio/config.conf` which makes GRC 3.8 crash on start:
```
Traceback (most recent call last):
  File "/home/schneider/gr-38/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/home/schneider/gr-38/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/home/schneider/gr-38/lib/python3/dist-packages/gnuradio/grc/main.py", line 83, in main
    platform.build_library()
  File "/home/schneider/gr-38/lib/python3/dist-packages/gnuradio/grc/core/platform.py", line 197, in build_library
    utils.hide_bokeh_gui_options_if_not_installed(self.blocks['options'])
  File "/usr/lib/python3.6/collections/__init__.py", line 883, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/lib/python3.6/collections/__init__.py", line 875, in __missing__
    raise KeyError(key)
KeyError: 'options'
```

Deleting `global_blocks_path` from `config.cfg` seems to make it work for both versions.

cpatulea, 2020-10-26:
Freshly installed 3.8.2.0 on Mac OS. I have *no* grc.conf:

```
$ ls -l ~/.gnuradio/grc.conf
ls: /Users/catalinp/.gnuradio/grc.conf: No such file or directory
```

and yet...
```
$ gnuradio-companion
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
Traceback (most recent call last):
  File "/usr/local/bin/gnuradio-companion", line 102, in <module>
    run_main()
  File "/usr/local/bin/gnuradio-companion", line 95, in run_main
    exit(main())
  File "/usr/local/Cellar/gnuradio/3.8.2.0_3/lib/python3.9/site-packages/gnuradio/grc/main.py", line 83, in main
    platform.build_library()
  File "/usr/local/Cellar/gnuradio/3.8.2.0_3/lib/python3.9/site-packages/gnuradio/grc/core/platform.py", line 197, in build_library
    utils.hide_bokeh_gui_options_if_not_installed(self.blocks['options'])
  File "/usr/local/Cellar/python@3.9/3.9.0/Frameworks/Python.framework/Versions/3.9/lib/python3.9/collections/__init__.py", line 941, in __getitem__
    return self.__missing__(key)            # support subclasses that define __missing__
  File "/usr/local/Cellar/python@3.9/3.9.0/Frameworks/Python.framework/Versions/3.9/lib/python3.9/collections/__init__.py", line 933, in __missing__
    raise KeyError(key)
KeyError: 'options'
```

cpatulea, 2020-10-26:
workaround worked for me:
```
$ pwd
/usr/local/Cellar/gnuradio/3.8.2.0_3/.bottle/etc/gnuradio/conf.d
$ cp -a grc.conf ~/.gnuradio/config.conf
$ gnuradio-companion
WARNING:gnuradio.grc.core.platform:Slow YAML loading (libyaml not available)
<<< Welcome to GNU Radio Companion 3.8.2.0 >>>

Block paths:
	/usr/local/Cellar/gnuradio/3.8.2.0_3/share/gnuradio/grc/blocks
```

justinh24, 2020-12-09:
I had this issue on a Homebrew (mac) install of gnuradio.  Reinstalling did now work.  This is due to the uninstall not removing the /usr/local/etc/gnuradio directory.  This folder had a config file that pointed to a folder that no longer existed from a previous release. 

Uninstalling, removing the /usr/local/etc/gnuradio, and  ~/.gnuradio folder, then reinstalling did the trick.

willcode, 2020-12-09:
I've started calling GRC from a shell script in each PyBOMBS directory that does:
`GR_CONF_GRC_GLOBAL_BLOCKS_PATH=$(PYBOMBS_PREFIX)/share/gnuradio/grc/blocks gnuradio-companion $@`
and multiple versions stay deconflicted.

marcusmueller, 2024-01-07:
since we've had our format change in 3.7->3.8, we don't have to ensure this compatibility any more. GRC currently does *not* crash when it encounters something in a YAML file it does not understand.

closed 2024-01-07 by marcusmueller